TheThingsNetwork/lorawan-stack #82: Unable to log in using ttn-lw-cli whilst following "getting started" guide

proffalken, 2019-02-01:
**Summary:** 
When trying to follow the "getting started" guide, all commands complete successfully until I run `docker-compose exec stack ttn-lw-cli login`.  This command produces the following output:
```
  WARN No access token present                 
  INFO Please go to https://localhost:8885/oauth/authorize?client_id=cli&response_type=code
  INFO Waiting for your authorization... 
```
and then waits for me to visit the URL.

When I visit the URL in a browser and accept the self-signed TLS certificate, I get redirected to `https://localhost:8885/oauth/login?n=%2Foauth%2Fauthorize%3Fclient_id%3Dcli%26response_type%3Dcode` which results in a blank screen.

Inspection of the javascript console reveals the following error:
```
Uncaught SyntaxError: Unexpected end of input  https://localhost:8885/assets/oauth.ed8f605040f0efbcfb3a.js:67
```
The client does not log in.

The logs in the console following the `docker-compose up` command are as follows:

```
[mmw@rincewind lorawan-stack]$ docker-compose up
lorawan-stack_redis_1 is up-to-date
lorawan-stack_cockroach_1 is up-to-date
Creating lorawan-stack_stack_1 ... done
Attaching to lorawan-stack_redis_1, lorawan-stack_cockroach_1, lorawan-stack_stack_1
redis_1      | 1:C 01 Feb 2019 05:20:22.129 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 01 Feb 2019 05:20:22.130 # Redis version=5.0.1, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1      | 1:C 01 Feb 2019 05:20:22.130 # Configuration loaded
redis_1      | 1:M 01 Feb 2019 05:20:22.131 * Running mode=standalone, port=6379.
redis_1      | 1:M 01 Feb 2019 05:20:22.131 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1      | 1:M 01 Feb 2019 05:20:22.131 # Server initialized
redis_1      | 1:M 01 Feb 2019 05:20:22.131 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
redis_1      | 1:M 01 Feb 2019 05:20:22.131 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
redis_1      | 1:M 01 Feb 2019 05:20:22.131 * DB loaded from append only file: 0.000 seconds
redis_1      | 1:M 01 Feb 2019 05:20:22.131 * Ready to accept connections
cockroach_1  | *
cockroach_1  | * WARNING: RUNNING IN INSECURE MODE!
cockroach_1  | * 
cockroach_1  | * - Your cluster is open for any client that can access <all your IP addresses>.
cockroach_1  | * - Any user, even root, can log in without providing a password.
cockroach_1  | * - Any user, connecting as root, can read or write any data in your cluster.
cockroach_1  | * - There is no network encryption nor authentication, and thus no confidentiality.
cockroach_1  | * 
cockroach_1  | * Check out how to secure your cluster: https://www.cockroachlabs.com/docs/v2.1/secure-a-cluster.html
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "a2dcbb6a2028" to other nodes, is this routable?
cockroach_1  | * 
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | * 
cockroach_1  | *
cockroach_1  | CockroachDB node starting at 2019-02-01 05:20:22.773655691 +0000 UTC (took 0.6s)
cockroach_1  | build:               CCL v2.1.1 @ 2018/11/19 18:24:21 (go1.10.3)
cockroach_1  | webui:               http://a2dcbb6a2028:26256
cockroach_1  | sql:                 postgresql://root@a2dcbb6a2028:26257?sslmode=disable
cockroach_1  | client flags:        /cockroach/cockroach <client cmd> --host=a2dcbb6a2028:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp659059884
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | status:              initialized new cluster
cockroach_1  | clusterID:           216a73d8-6d31-4bb8-8ca1-3ebbcfb2b626
cockroach_1  | nodeID:              1
stack_1      |   INFO Setting up core component               
stack_1      |   WARN No cookie hash key configured, generated a random one hash_key=5CC7620C17255B463FF6ACB48596BF036BDD54BF714AFBFA8ADF503874E9AD8504F5A595DF38187DD22CB6054906EB27B29FD7EEB374BB6707169FCB3194656D namespace=web
stack_1      |   WARN No cookie block key configured, generated a random one block_key=FF08EB829C513E72990DEB5E3BAAC1945ACECF173C3FE471B489B70D196D769E namespace=web
stack_1      |   INFO Setting up Identity Server              
[... 51 lines cut ...]
  API version:      1.38 (minimum version 1.12)
  Go version:       go1.11
  Git commit:       e68fc7a215
  Built:            Fri Sep  7 11:26:11 2018
  OS/Arch:          linux/amd64
  Experimental:     false
```

OS: Arch Linux

Browser: 
```

Google Chrome | 62.0.3202.94 (Official Build) (64-bit)
-- | --
Revision | 4fd852a98d66564c88736c017b0a0b0478e885ad-refs/branch-heads/3202@{#789}
OS | Linux
JavaScript | V8 6.2.414.42
Flash | 32.0.0.114 /home/mmw/.config/google-chrome/PepperFlash/32.0.0.114/libpepflashplayer.so
User Agent | Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/62.0.3202.94 Safari/537.36
Command Line | /opt/google/chrome/google-chrome --flag-switches-begin --flag-switches-end

```

**What can you do yourself and what do you need help with?**

proffalken, 2019-02-01:
With the help of @bafonins @kschiffer @rvolosatovs we have tracked this issue back to 3bf5834 

We can confirm that 2d94569 works as expected.

closed 2019-02-01 by johanstokking

nankeen, 2019-02-13:
Hi I'm experiencing a similar issues. Following the guide I was able to reach a web login page. However after submitting my credentials there was a status 500 response giving the following on a blank screen:
```
HTTP error: urls cannot be blank.
```

PR #83 (2019-02-01): Revert add Echo HTTP Middleware
**Summary:**

Closes #82 

**Changes:**

Revert commit 3bf5834ae21ae597ed8db6a26121218e4512c65c.