TheThingsNetwork/lorawan-stack #84: Add Echo HTTP middleware

johanstokking, 2019-02-01:
**Summary:** 

Implement the Echo middleware as in https://github.com/TheThingsNetwork/lorawan-stack/commit/3bf5834ae21ae597ed8db6a26121218e4512c65c, but without breaking the login flow as reported in #82.

**Why do we need this?**

Need request body limit, need compression, need some security headers and recovery from panics.

**What is already there? What do you see now?**

Reverted the functionality described above because it broke the login flow as described in #82.

**What is missing? What do you want to see?**

- A request body limit of 16 MB
- Gzip compression
- Security headers, such as
  ```
  X-Content-Type-Options: nosniff
  X-Frame-Options: SAMEORIGIN
  X-Request-Id: 01D2FQ0R758RZ4X99NPHYV44AW
  X-Xss-Protection: 1; mode=block
  ```
- Recovery from panics

Additionally, it adds CORS headers to all `/api/v3` endpoints, as those may be called by external consoles.

bafonins, 2019-02-04:
After a quick investigation, it turned out that `middleware.Gzip()` is the one causing the issue. Apparently, there is something wrong with the `Content-Length` header set by the middleware which causes the browser to process js/css files of invalid length.

johanstokking, 2019-02-04:
@bafonins happy to see a PR without that middleware and push gzip to another PR.

closed 2019-02-04 by bafonins