TheThingsNetwork/lorawan-stack #85: nats support

FlorianLudwig, 2019-02-01:
**Summary:** 

Integrate [NATS](https://nats.io/) as alternative to pubsub for applications to consume packages.
...

**Why do we need this?**

Proper queuing & load balancing.

**What is already there? What do you see now?**

 * Pubsub MQTT
 * Webhook

**What is missing? What do you want to see?**

The webhook has no queuing / retry (especially no disk based since we don't want to run out of ram when running with a lot of devices).

**How do you propose to implement this?**

A NATS instance / cluster should be configurable for the application server.

@johanstokking seems to have an implementation already in mind (I just stole his idea and turned it into a issue report)

**What can you do yourself and what do you need help with?**

I am getting familiar with the code base but are way off to proving a sane implementation.  So best thing I can do right now is being a pain in the neck to Johan by whining how nice it would be to have proper queuing.

johanstokking, 2019-02-06:
There are actually two ways we can implement this; a hosted NATS server, like we host MQTT in the AS now, or a NATS publisher to a standalone NATS server.

I'm assuming the latter approach is preferred here. This allows for NATS clustering, persistence, custom authentication, configurable ports etc.

htdvisser, 2019-02-11:
If we do the latter, it would probably also be a good idea to add a NATS backend for the events package as well.

johanstokking, 2019-02-11:
Will implement it with that in mind.

johanstokking, 2019-05-14:
@adriansmares let's discuss the concerns for `go-cloud` support here.

Regarding the metadata; please check if we can use this. Note that we still need to support `[]ttnpb.ApplicationDownlink` when replacing the queue. So we already don't pubsub loose `ttnpb.ApplicationDownlink` messages.

adriansmares, 2019-05-14:
I've tested pushing uplinks both with, and without metadata, and the results are as follows:
* Pushing messages only with a body, without any metadata, results in just the body being pushed:
```
[#1] Received on [testtopic] : '{"end_device_ids":{"device_id":"dev1","application_ids":{"application_id":"app1"},"dev_eui":"0004A30B001C0530","join_eui":"800000000000000C","dev_addr":"01A34BDC"},"correlation_ids":["as:up:01DAVAY90K2MJ1JVTXD1NDM4Y7","gs:conn:01DAVAY4166FKNG5PWEB5ZN7RT","gs:uplink:01DAVAY8T0X59VNVEFKH0RGW08","ns:uplink:01DAVAY8T2A437BF9PCR0SBKEW","rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01DAVAY8T2QW8EC7HP9JS6Q055"],"received_at":"2019-05-14T14:11:32.290238381Z","join_accept":{"session_key_id":"AWq2ryNFyiIhIMK2cYJk1Q=="}}'
```

* Pushing both the body and the metadata results in a binary format [(GOB for NATS)](https://github.com/google/go-cloud/blob/master/pubsub/natspubsub/nats.go#L449):
```
[#2] Received on [testtopic] : '��Body��{"end_device_ids":{"device_id":"dev1","application_ids":{"application_id":"app1"},"dev_eui":"0004A30B001C0530","join_eui":"800000000000000C","dev_addr":"010E4899"},"correlation_ids":["as:up:01DAVB3WD09HHGEX324CXT4EYY","gs:conn:01DAVB2HN5FSJEEZZ7AEA66VWG","gs:uplink:01DAVB3W5GKE4C0W48Y4ARPPXY","ns:uplink:01DAVB3W5JXE1KPDJP8N27J77J","rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01DAVB3W5H1RV5EZNKHKZGTF9S"],"received_at":"2019-05-14T14:14:35.954371970Z","join_accept":{"session_key_id":"AWq2sfC1W+kj+aeWLOuEMQ=="}}�Metadata��meta1�value1�meta2�value2'
```

So I'd suggest we'd rather use only the body, in JSON format, or at most protobuf, in order to allow non-Go clients to consume what we publish. Also regarding slices: why would that be an issue ? They can stay as JSON arrays / repeated structures in Protobuf.

johanstokking, 2019-05-14:
Great. Yeah indeed JSON it is (no protobuf); the point here is interoperability with developer applications which are most often not using `go-cloud`.

With the slices, indeed this would be an array, but the `ApplicationDownlink` does not contain the operation (push/replace) nor the end device identifiers. So even if we would be using metadata for that (which we can't), these would not be loose messages like `ApplicationUp`. You may consider embedding `DownlinkQueueRequest` and adding the operation as enum there.

adriansmares, 2019-05-14:
Since we cannot break the API, I've created an additional `DownlinkQueueOperation` message for this purpose. I'll update the PR with the new implementation and details.

johanstokking, 2019-05-14:
You don't need to break API if you introduce that message that embeds `DownlinkQueueRequest`, but I'm fine with either.

PR #673 (2019-05-28): Application Server PubSub frontend

#### Summary

Closes #85

#### Changes

- Added the `go-cloud` pubsub frontend to the application server.

#### Notes for Reviewers

As discussed in #85, no metadata is being sent in order to ensure developer friendliness.

One can test the integration, for example, by first starting a NATS server:
```
# Install the default NATS server
go get github.com/nats-io/nats-server
nats-server
```

Then starting the stack
```
export NATS_SERVER_URL="nats://localhost"
ttn-lw-stack start --as.pubsub.publish-url "nats://pubtopic" --as.pubsub.subscribe-url "nats://subtop [... cut]
closed 2019-05-28 by adriansmares