TheThingsNetwork/lorawan-stack #88: Documentation is unclear about how to authenticate client when not working on the "localhost"

proffalken, 2019-02-03:
**Summary:** 
When running the v3 stack on a remote host, attempting to authenticate the client against that remote host fails with a TLS error in the lora-stack logs.

**Steps to Reproduce:** 
1. Run the stack on a system other than your local machine
2. From your local machine, attempt to run `ttn-lw-cli login` with the appropriate flags to point to the remote server, for example:
```
go run ./cmd/ttn-lw-cli login  \
      --application-server-address="ttnv3.mockingbirdconsulting.co.uk:8884" \
                              --ca="" \
          --gateway-server-address="ttnv3.mockingbirdconsulting.co.uk:8884" \
         --identity-server-address="ttnv3.mockingbirdconsulting.co.uk:8884" \
                    --input-format="json" \
                        --insecure="false" \
             --join-server-address="ttnv3.mockingbirdconsulting.co.uk:8884" \
                       --log.level="info" \
          --network-server-address="ttnv3.mockingbirdconsulting.co.uk:8884" \
            --oauth-server-address="https://ttnv3.mockingbirdconsulting.co.uk:8885" \
                   --output-format="json" \
``` 
3. Access the URL displayed by the client for oauth authentication
4. Log in to the console

**What do you see now?**
In the browser: `Could not exchange OAuth access token`

In the cli output:
```
[ttnv3@ttnv3demo lorawan-stack]$ ./client.sh login
  WARN No access token present                 
  INFO Please go to https://ttnv3.mockingbirdconsulting.co.uk:8885/oauth/authorize?client_id=cli&response_type=code
  INFO Waiting for your authorization...       
 ERROR Could not exchange OAuth access token    error=Post https://ttnv3.mockingbirdconsulting.co.uk:8885/oauth/token: x509: certificate signed by unknown authority
```

In the `lora-stack` logs:

```
2019/02/03 12:14:14 http: TLS handshake error from 167.99.199.54:35776: remote error: tls: bad certificate
```

This is a temporary server setup for testing purposes, so I'm happy to provide access if desired.

ttnv3.mockingbirdconsulting.co.uk resolves to 167.99.199.54, and this does not work either on my laptop on locally on the server.

The Certificates are LetsEncrypt and the certificates are valid for the browser.

**What do you want to see instead?**
Documentation that shows how to access and log in to the console from a host other than "localhost".

**How do you propose to implement this?**
PR against the docs once I understand how it all works!

**Environment:**
Centos 7, lorawan-stack running directly from disk, no Docker.

**What can you do yourself and what do you need help with?**
If you can explain how to do it, I can write it up :)

rvolosatovs, 2019-02-04:
I set up the stack on a remote server and managed to login via CLI using the Let's Encrypt certificates.
Note, that I ran the CLI on the remote server, *not* on my local machine.
What did you specify as the callback URL when registering the CLI oauth client?

proffalken, 2019-02-04:
I can get it working if I run everything on the remote machine, it's when I try and run the cli on my laptop with the server remotely that it fails.

How does the cli work for the oAuth? Does is set up it's own webserver for the callback? If so, that would explain it as I'd need to open up my local firewall to allow the traffic.

The callback URL was the test server when running this command

rvolosatovs, 2019-02-04:
CLI indeed acts as "it's own webserver", so in order to login at the remote machine from your local one, the callback you register when creating the CLI OAuth client would need to point to that local machine.
Another(untested) solution could be setting up an SSH tunnel, for example, forwarding traffic on `:11885` from your remote machine to the local one,  that way you would not need to change the URL.

johanstokking, 2019-02-04:
@proffalken we're opening another issue for this. We need to have a way for the cli to obtain an access code without the callback, or potentially a callback to a page from which you can copy something to the cli input.

Is your specific issue here still relevant then?

rvolosatovs, 2019-02-05:
See https://github.com/TheThingsNetwork/lorawan-stack/issues/99