TheThingsNetwork/lorawan-stack #95: IntlHelmet crashes on oauth login page

bafonins, 2019-02-04:
**Summary:** 
`<IntlHelmet />` on the oauth [(login page)](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/views/oauth/login/index.js#L81-L83) crashes the app when submitting the form with invalid credentials.
Note, that the issue dissapears when the component is commented out.

**Steps to Reproduce:** 

1. go to `/oauth/login`
2. submit the form with invalid credentials
3. see the console

**What do you see now?**

`Uncaught RangeError: Maximum call stack size exceeded`

**What do you want to see instead?**

Normal behaviour without crashes.

**How do you propose to implement this?**

The problem might be that the `<IntlHelmet />` returns a new component identical to the one before rerender that forces `react-helmet` to examine every property of the children.

**Environment:**

Browser

**What can you do yourself and what do you need help with?**
Spend some time debugging the issue and it turns out that the problem lies here: https://github.com/nfl/react-helmet/blob/master/src/Helmet.js#L93 . When the login page rerenders after showing the error `react-helmet` falls into infinite(?) recursion trying to compare the children of `<IntlHelmet />` which are react nodes having quite some properties where most of them are nested.

kschiffer, 2019-02-05:
I can't reproduce this issue. Tried both on `production` and `development`. Is there maybe a precondition that you've missed?

bafonins, 2019-02-05:
@kschiffer try submitting the empty form