TheThingsNetwork/lorawan-stack #103: Add support for updating/deleting users and OAuth clients in stack

rvolosatovs, 2019-02-05:
**Summary:** 

Add support for updating/deleting users and OAuth clients in stack

**Why do we need this?**

To be able to modify setup without having to manually edit the database and/or wiping the data

**What is already there? What do you see now?**

Only create functionality

**What is missing? What do you want to see?**

Update and delete functionality. Maybe also list.

**How do you propose to implement this?**

Add the commands

htdvisser, 2019-02-11:
Could you be more clear about what exactly is missing? We already have these commands in the CLI:

```
% go run ./cmd/ttn-lw-cli cli -h
Client commands

Usage:
  ttn-lw-cli clients [command]

Aliases:
  clients, client, cli, c

Available Commands:
  collaborators Manage client collaborators
  contact-info  Manage client contact info
  create        Create a client
  delete        Delete a client
  get           Get a client
  list          List clients
  rights        List the rights to a client
  search        Search for clients
  update        Update a client
```

and for users:

```
% go run ./cmd/ttn-lw-cli usr -h
User commands

Usage:
  ttn-lw-cli users [command]

Aliases:
  users, user, usr, u

Available Commands:
  api-keys        Manage user API keys
  contact-info    Manage user contact info
  create          Create a user
  delete          Delete a user
  get             Get a user
  invitations     Manage user invitations
  rights          List the rights to a user
  search          Search for users
  update          Update a user
  update-password Update a user password
```

Listing users is currently indeed not possible, but as this can only be done by admins anyway, we can use search for that.

rvolosatovs, 2019-02-11:
Example from https://github.com/TheThingsNetwork/lorawan-stack/issues/102:
```
Usage:
  ttn-lw-stack oauth [command]

Available Commands:
  create Create an OAuth client
  update Update an OAuth client
  delete Delete an OAuth client
```

The issue is e.g. that after doing `ttn-lw-stack is-db create-oauth-client --id cli --name "Command Line Interface" --owner admin --no-secret --redirect-uri "X"` there seem to be no way to change/delete the client without either wiping the IS database or manually manipulating it.

htdvisser, 2019-02-11:
As @johanstokking wrote in that issue, the commands in `ttn-lw-stack` are for bootstrapping only. After running the `ttn-lw-stack is-db create-oauth-client` command you mention, the `admin` user can use `ttn-lw-cli` to manage the client.

closed 2019-02-11 by htdvisser