TheThingsNetwork/lorawan-stack #104: update or create OTAA devices and specify DevAddr, AppSKey, NetSKey, FcntUp, FnctDown

pdubois22, 2019-02-05:
**Summary:** 
ability to update or create OTAA devices and specify DevAddr, AppSKey, NetSKey, FcntUp, FnctDown in addition to DevEUI, JoinEUI and AppKey using the cli

**Why do we need this?**
We need to transfer existing devices that have already joined other Network Servers (ttnv2 and others) to ttnv3. Rejoining is not an option since those devices are on the field. We have session keys for those devices and up / down counters

**What is already there? What do you see now?**
ttn-lw-cli devices update and create are there, but the option --session.dev_addr= does not seem to update the registry. I still got "WARN No device matched DevAddr". In addition, --session.keys.app-s-key.key does not work (value not saved). bug?

**What is missing? What do you want to see?**
ttn-lw-cli devices update work as we have with ttnv2: 
ttnctl devices set xxx --override --dev-addr xxx --nwk-s-key xxx --app-s-key xxx --fcnt-up  nnn --fcnt-down  nnn

**How do you propose to implement this?**
not sure. is updating the session the right thing to do?

**Environment:**
lorawan-stack on a linux server.

**What can you do yourself and what do you need help with?**
Test any fix or workaround or try to fix with guidance

johanstokking, 2019-02-05:
Agreed, we should support this. In fact, I think we do, but you have to disable defaults and specify all settings yourself.

Here's an example:

```
$ ttn-lw-cli devices create test migrated --defaults=false --supports-join=true --join-eui=0001020304050607 --dev-eui=0001020304050607 --supports-class-b=false --supports-class-c=false --root-keys.root-key-id=migrated --root-keys.app-key.key=000102030405060708090a0b0c0d0e0f --session.dev-addr=01020304 --session.keys.app-s-key.key=000102030405060708090a0b0c0d0e0f --session.keys.f-nwk-s-int-key.key=000102030405060708090a0b0c0d0e0f --lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b --frequency-plan-id EU_863_870 --mac-settings.use-adr=true --resets-f-cnt=false --resets-join-nonces=false --uses-32-bit-f-cnt=true
```

Please list steps to reproduce in order to pin down this issue.

pdubois22, 2019-02-06:
nope, not working.
attached steps to reproduced and debug info. Let me know if you need more

device is not matched because MacState is nil:

pkg/networkserver/grpc_gsns.go:

	if err := ns.devices.RangeByAddr(pld.DevAddr,
		[]string{
			"frequency_plan_id",
			"lorawan_phy_version",
			"mac_state",
			"pending_session",
			"recent_downlinks",
			"recent_uplinks",
			"resets_f_cnt",
			"session",
			"uses_32_bit_f_cnt",
		},
		func(dev *ttnpb.EndDevice) bool {
			if dev.MACState == nil {         // <--- dev.MACState is nil. Device not matched
				return true
			}

[lorawan-stack-issue104.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/2835739/lorawan-stack-issue104.txt)

rvolosatovs, 2019-02-06:
An easy workaround would be to register the device with `supports-join=false`, however that would mean that NS will consider the device as ABP and so the joins will never succeed.
Once/if rejoining becomes possible, you could reregister the device as OTAA. Would that be an acceptable option?

pdubois22, 2019-02-06:
not a good option. if the device try to rejoin for whatever reason (power loss, battery removed, reset with magnet, ...) the ns will ignore it and the device will be lost.
i think we should support it, otherwise, we have a migration issue to ttnv3. if you have another solution to migrate devices, or a sequence that works (create then update?), or that may work, I am more than happy to test it.

rvolosatovs, 2019-02-06:
You should be able to create the device as ABP and then simply update it to OTAA (by changing the `supports-join` from `false` to `true`)

pdubois22, 2019-02-06:
I may be doing something stupid:

when I try to register the existing device with ABP:

go run ./cmd/ttn-lw-cli --ca="/home/thethings/lorawan-stack/cert.pem" devices create test migrated \
--defaults=true \
--supports-join=false \
--dev-eui=A81758FFFE037503 \
--session.dev-addr=FC0069C1 \
--session.keys.app-s-key.key=9F264F712940CA819A4D19EA91B762EB \
--session.keys.f-nwk-s-int-key.key=E8E2B56D2114E8A0DA0C4338A008DB70 \
--lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
--frequency-plan-id EU_863_870

I got sometimes:
ERROR Failed to update device in registry      dev_addr=FC0069C1 device_uid=test.migrated error=error:pkg/networkserver:uplink_channel_not_found (uplink channel not found) grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D31F3PNRM6J4ECEM7SZCKK5E

or worse, the entire ns crashes:
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x1513b0c]

goroutine 206 [running]:
go.thethings.network/lorawan-stack/pkg/networkserver.generateDownlink(0x1c94240, 0xc000f89d70, 0xc000c01b00, 0x7b001d, 0xc00017e070, 0x1c94240, 0xc000f89d10, 0x3, 0x0, 0xc0002ca560, ...)
	/home/thethings/lorawan-stack/pkg/networkserver/downlink.go:167 +0xafc
go.thethings.network/lorawan-stack/pkg/networkserver.(*NetworkServer).processDownlinkTask.func1.1(0xc000c01b00, 0xc000c018c0, 0xc00096c7e0, 0x9, 0x9, 0xc000c01b00, 0x0)
	/home/thethings/lorawan-stack/pkg/networkserver/downlink.go:593 +0x6b6
go.thethings.network/lorawan-stack/pkg/networkserver/redis.(*DeviceRegistry).SetByID.func1(0xc000267440, 0xc000a532a8, 0x1)
	/home/thethings/lorawan-stack/pkg/networkserver/redis/registry.go:165 +0x285

see attached file
[lorawanstack-issue104.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/2836477/lorawanstack-issue104.txt)

johanstokking, 2019-02-06:
> You should be able to create the device as ABP and then simply update it to OTAA (by changing the `supports-join` from `false` to `true`)

This is not really what we want. We need to be able to migrate the device state of a joined device through API and CLI.

pdubois22, 2019-02-19:
#101 does not close this one.
uplink is ignored due to device not matched.

because MacState is nil in this case since join request was never received.

rvolosatovs, 2019-02-19:
@pdubois22 you should specify:
`--session.dev_addr`
`--session.keys.f_nwk_s_int_key.key`
During *creation* of your pre-LoRaWAN1.1 device.
Note, that I just added a small commit, which should eliminate any possible issues 293800281920526874720a3360cba09a550af83e
Care to try again?

pdubois22, 2019-02-19:
@rvolosatovs : it does not work:

go run ./cmd/ttn-lw-cli devices create test dev2 --defaults=true \
--supports-join=true \
--join-eui=70B3D59520000000 --dev-eui=A81758FFFE032277 \
--root-keys.app-key.key=4F939595CE83288AFEF81BD881C3C36E \
--session.dev-addr=010B600C \
--session.keys.app-s-key.key=A15BEF4A3233274AE917E4AFB19055F2 \
--session.keys.f-nwk-s-int-key.key=9E2FB61D7310C9279886DB79FA52F9F4 \
--lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
--frequency-plan-id EU_863_870 \
--mac-settings.use-adr=true \
--resets-f-cnt=false \
--resets-join-nonces=false \
--session.last-f-cnt-up=45872 \
--session.last-n-f-cnt-down=1880 \
--uses-32-bit-f-cnt=true

=>

DEBUG Deduplicating uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D42M1FYPQBCBVQ8SA0HQR391
 DEBUG Handling data uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D42M1FYPQBCBVQ8SA0HQR391
 DEBUG Matching device...                       dev_addr=010B600C grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D42M1FYPQBCBVQ8SA0HQR391
  WARN No device matched DevAddr                dev_addr=010B600C grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc payload_length=26 request_id=01D42M1FYPQBCBVQ8SA0HQR391 uplink_f_cnt=46470
  WARN Failed to match device                   dev_addr=010B600C error=error:pkg/networkserver:device_not_found (device not found) grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D42M1FYPQBCBVQ8SA0HQR391
 DEBUG Waiting for collection window to be closed... grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D42M1FYPQBCBVQ8SA0HQR391

Probable cause:

grpc_gsns.go l 117 : dev.MACState is nil.

			if dev.MACState == nil {
 				return true
 			}

rvolosatovs, 2019-02-19:
Please try this commit: https://github.com/TheThingsNetwork/lorawan-stack/pull/101/commits/8d74dfa115fec1aefb6b8d4afb9b2a3468a96c2d
I added a testcase based on your comment as well

pdubois22, 2019-02-19:
device is matched which is good, but still got a panic:

any clue ?

 DEBUG Handling data uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D43JBPG7C2ZJ37KA3TDKAGD8
 DEBUG Matching device...                       dev_addr=010B600C grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D43JBPG7C2ZJ37KA3TDKAGD8
 DEBUG Performing MIC checks on devices with matching frame counters... dev_addr=010B600C device_count=1 device_uid=test.dev2 full_f_cnt_up=46576 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs last_f_cnt_up=46575 namespace=grpc payload_length=26 request_id=01D43JBPG7C2ZJ37KA3TDKAGD8 uplink_f_cnt=46576 uplink_f_cnt_up=46576
 DEBUG Matched device                           dev_addr=010B600C device_uid=test.dev2 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D43JBPG7C2ZJ37KA3TDKAGD8

  WARN Application Server not found, not forwarding uplink dev_addr=010B600C device_uid=test.dev2 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D43JBPG7C2ZJ37KA3TDKAGD8
 DEBUG Waiting for collection window to be closed... grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D43JBPG7C2ZJ37KA3TDKAGD8
 DEBUG Processing downlink task...              delay=1.842ms device_uid=test.dev2 start_at=2019-02-19 19:05:15.21814 +0000 UTC
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x216b11e]

goroutine 227 [running]:
go.thethings.network/lorawan-stack/pkg/networkserver.(*NetworkServer).generateDownlink(0xc000870000, 0x291d640, 0xc000757800, 0xc000240fc0, 0xe60019, 0x291d640, 0xc0007577a0, 0x3, 0x0, 0xc00023a580, ...)
	/Users/pdubois/lorawan-stack/pkg/networkserver/downlink.go:237 +0xf6e
go.thethings.network/lorawan-stack/pkg/networkserver.(*NetworkServer).processDownlinkTask.func1.1(0xc000240fc0, 0xc000240d80, 0xc000974ab0, 0x9, 0x9, 0xc000240fc0, 0x0)
	/Users/pdubois/lorawan-stack/pkg/networkserver/downlink.go:617 +0x6ac
go.thethings.network/lorawan-stack/pkg/networkserver/redis.(*DeviceRegistry).SetByID.func1(0xc000a40ea0, 0xc000a692a8, 0x1)
	/Users/pdubois/lorawan-stack/pkg/networkserver/redis/registry.go:165 +0x285
github.com/go-redis/redis.(*Client).Watch(0xc00004bf80, 0xc000a69358, 0xc000a692a8, 0x1, 0x1, 0x1b, 0x254f4a0)
	/Users/pdubois/go/pkg/mod/github.com/go-redis/redis@v6.15.1+incompatible/tx.go:45 +0x51
go.thethings.network/lorawan-stack/pkg/networkserver/redis.(*DeviceRegistry).SetByID(0xc000236af0, 0x291d640, 0xc000757290, 0xc000264420, 0x4, 0xc000000000, 0xc000264425, 0x4, 0xc000974ab0, 0x9, ...)

rvolosatovs, 2019-02-19:
https://github.com/TheThingsNetwork/lorawan-stack/pull/101/commits/db4a24e406f4e98d97414b634ff48dde1621857c should fix it

rvolosatovs, 2019-02-19:
FTR, this is the most recent commit in the tree, where this issue should be fixed: https://github.com/TheThingsNetwork/lorawan-stack/commit/5127c2ccfa0aebaedadd600cc13d58088290774b
You can also just use #101 directly (`curl -sL https://github.com/TheThingsNetwork/lorawan-stack/pull/101.patch | git am`) on latest `master`

pdubois22, 2019-02-20:
@rvolosatovs : confirmed fixed with #101. no more panic. uplink and downlink handled gracefully.

I will do additional tests, but it looks good.

many thanks.

rvolosatovs, 2019-02-20:
That's great to hear, you're most welcome.
Thank you for testing this!

johanstokking, 2019-02-20:
@pdubois22 there will be a breaking change in the database, API and CLI flags, very unfortunately, in the `3.0.0` release. Please do not migrate production devices to a V3 stack just yet; wait for `3.0.0`.

PR #101 (2019-02-22): Overriding desired MAC parameters and defaults
**Summary:**

References #5 
Closes #60 
Closes #104 
Closes #105 
Closes #144 
Closes #159 

**Changes:**

- Add all optional fields from Backend Interfaces `DeviceProfile` as well as additional MAC parameters to `MACSettings`
- Ensure device API is aligned with `DeviceProfile ` from Backend Interfaces
- Make `MACSettings` optional
- Allow specifying NS-wide default `MACSettings` to fallback to if not defined by the device
- Update protoc image version (requires https://github.com/TheThingsIndustries/protoc-gen-fieldmask/pull/20)
- Set `RxDelay`  [... cut]
closed 2019-02-22 by rvolosatovs