TheThingsNetwork/lorawan-stack #107: Split oauth and console apps

bafonins, 2019-02-05:
**Summary:** 
Consider splitting the oauth and console apps within the `webui` package.

**Why do we need this?**
At the moment both apps reside in the same directory and reuse most of the functionality from each other, e.g. reducers, actions, hocs . This shouldnt be the case.
Such an approach makes it difficult to maintain both apps and tune similar functionality when needed. For example, there absolutely no need to initialize all the reducers from console, when only using the oauth app. Also, the way to determine the current user is different, as well as the server responses to fetch user entity. 
Besides that, it also makes it difficult to make use of the js sdk in the shared api entry, since it should be initialized only after successful authentication because only at that point the console can find out what the stack endpoints are. 

**What is already there? What do you see now?**
Single reducer, reusable actions, single api entry.

**What is missing? What do you want to see?**
Reducer, actions, api per app.

**How do you propose to implement this?**
Split the `webui` package into something like:

webui
├ console
├ oauth
├ ui/component
├ lib
├ console.js
├ oauth.js

**What can you do yourself and what do you need help with?**
Restructure the `webui` package

cc @kschiffer

kschiffer, 2019-02-05:
> Besides that, it also makes it difficult to make use of the js sdk in the shared api entry, since it should be initialized only after successful authentication because only at that point the console can find out what the stack endpoints are.

This is not actually the case. Other than that, yes, we should split this more clearly than we do now.

bafonins, 2019-02-05:
> This is not actually the case.

Actually, keeping all (oauth/console) api's together causes problem when trying to initialize the sdk in the oauth app, since there is no stack configuration injected.

bafonins, 2019-02-06:
Let me identify the main problems related to this issue:
1. The oauth and console api definitions are currently stored together. This makes it difficult to initialize the js sdk in there for use in the console, since it will crash because of the stack configuration absence in the oauth app template. At the moment I see only 3 ways to solve this:
    * Initialize the sdk as a thunk. This forces us to reinitialize the sdk on every use, meaning that we also have to generate all api methods from scratch.
    * Adding check for the stack configuration to the sdk. This solves the issues, however solution seems quite dirty.
    * Simply split `webui/api/index.js` into `webui/api/console.js` and `webui/api/oauth.js`. This is the cleanest solution, imo.
2. Reuse of the `<Init />` ([init.js](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/lib/components/init.js)) component that forces us to reuse some redux primitives as well. Because of that we get [this](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/logic/init.js#L31-L43). This approach overall is error-prone. Especially, taking into account that there might the error added in the future.
    * We can split the `<Init />` component for each app, also move the global definitions, e.g. `react-grid-system` to the apps. Each component will dispatch appropriate actions and perfom the initialization logic for the app it was created for. There are various ways to implement it.
3. Redundancy in the redux store. Currently, the whole store is initialized with the console related entities while using the oauth app and some reducers are reused, e.g. the `user` reducer.
    * We can make the store for each app and initialzing these with appropriate root reducer. I see it as having `store/console.js`, `reducers/console.js` and the same of the oauth app.
    * Make use of [replaceReducer](https://redux.js.org/api/store#replaceReducer) ?
4. The overall threatment of both apps as being mandatory tied together. I had this problem of identifying responsibilities of each app initially. They are completely separated, semantically different and can be deployed separately. 
    * Well, split them?

After clearly defining the problems with the current structure, the solution taken in https://github.com/TheThingsNetwork/lorawan-stack/pull/106 seems to be overkill. At the moment every issue can be solved without such massive changes for the whole `webui` package. There might be problems with that in the future, however now I cant come up with one.

Let me know what you think, mb I am missing something.

closed 2019-02-11 by bafonins

kschiffer, 2019-02-11:
Just for completeness, I want to add my opinion here, as we discussed on slack:
I think splitting the webui package is a chance to be taken now as the codebase is still relatively small and it will enable us a better separation of concerns. We realized the two apps only commonality is the components. Splitting the apps more clearly will hence help us regarding them properly isolated.