TheThingsNetwork/lorawan-stack #99: Support logging in to a remote stack from local CLI

rvolosatovs, 2019-02-05:
**Summary:** 

Logging into a remotely-deployed stack from  locally-running CLI should be supported.
Also see #88
cc @johanstokking 

**Why do we need this?**

Users should be able to use CLI locally to manage their devices, applications and whatnot in TTN

**What is already there? What do you see now?**

https://github.com/TheThingsNetwork/lorawan-stack/blob/257f2821768159b0210a981eae6fb540854d6c5f/cmd/ttn-lw-cli/commands/login.go#L33-L73
Support for logins when server and CLI are running on the same host

**What is missing? What do you want to see?**

Support for remote logins

**How do you propose to implement this?**

Show a token during login in the browser, which the user can copy

**Environment:**

Any

rvolosatovs, 2019-02-05:
@proffalken I updated the issue - I was wrong, the remote logins are already supported for callback URL `http://localhost:11885/oauth/callback`

pdubois22, 2019-02-05:
the call back to localhost may need the certificate or the ca certificate to validate the return token: go run ./cmd/ttn-lw-cli login --ca="/home/thethings/lorawan-stack/cert.pem" ...

johanstokking, 2019-02-11:
Looping in @bafonins because I'm quite sure we need some subtle frontend engineering here.

htdvisser, 2019-02-11:
We could implement this as follows:

1. The OAuth Client for the CLI will be registered in the IS with an extra whitelisted redirect URI: `<OAuthServerAddress>/oauth/code`
2. We add a `--copy-code` flag to the CLI that sets the redirect URI in the authorize link to `<OAuthServerAddress>/oauth/code`
3. We add a little frontend page on `/oauth/code` that displays the `code` from the query and tells the user to copy-paste it into the CLI. I think we no longer need the `state` param.
4. The CLI will interactively take the authorization code and do the exchange as usual.

htdvisser, 2019-02-11:
In progress: https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/99-copy-paste-oauth

johanstokking, 2019-02-11:
@htdvisser shouldn't the `--copy-code` flag be default? I think this behavior is expected.

rvolosatovs, 2019-02-11:
Can we do both?
So try sending the request from frontend to CLI first and if that fails - show the code to copy-paste.
The CLI simply always reads stdin after `login` and if the request arrives from frontend - proceeds immediately

johanstokking, 2019-02-12:
That would make it time-dependent, and how would the frontend need to deal with this? The browser doesn't know if the redirect works.

Let's continue the discussion in #128. I made another suggestion there.

closed 2019-02-12 by htdvisser

Oliv4945, 2019-02-15:
> @htdvisser shouldn't the `--copy-code` flag be default? I think this behavior is expected.

I second that

PR #128 (2019-02-12): Add OAuth flow with Authorization Code copy-paste for CLI
**Summary:**

This PR is intended to resolve #99 by letting users disable the CLI's OAuth callback endpoint with `--callback=false`, which will make the OAuth flow redirect to a new page on the OAuth server that simply displays the authorization code for copy-paste.

**Changes:**

Added a `--callback=false` flag to the CLI that changes the Redirect URI and asks for the authorization code:

[image]

Added web UI that allows users to copy the authorization code:

[image]

And finish the OAuth flow with the usual token exchange:

[image]

 [... cut]