TheThingsNetwork/lorawan-stack #110: Add AUR package

rvolosatovs, 2019-02-07:
**Summary:** 

Package TTN stack and CLI for AUR

**Why do we need this?**

To make users happy

**What is already there? What do you see now?**

`goreleaser` setup once #109 is merged

**What is missing? What do you want to see?**

AUR package.

**How do you propose to implement this?**

We could wait until https://github.com/goreleaser/goreleaser/issues/641 is done, so the process could be automated

**Environment:**

CI

htdvisser, 2021-03-26:
Is there any demand for this? Otherwise I think we should just close this.

rvolosatovs, 2021-03-26:
Same is true for e.g. brew package, right?

htdvisser, 2021-03-26:
Looks like we had [around 95 Homebrew installs in the last 90 days](https://formulae.brew.sh/analytics/install-on-request/90d/), so that's definitely used by some people.

rvolosatovs, 2021-03-26:
Great, and the fact that we already have a package is the prerequisite for that.
We have 0 v3 AUR installs because we do not have an v3 AUR package.
There are even 2 community-contributed v2 packages though:
- https://aur.archlinux.org/packages/ttnctl/
- https://aur.archlinux.org/packages/ttnctl-bin/

htdvisser, 2021-03-30:
As just discussed, we'll keep the issue open, but don't work on it unless it gets some upvotes.

KrishnaIyer, 2025-04-30:
We have no requests for this and since Goreleaser is already error prone with Snapcraft, let's not do this.

closed 2025-04-30 by KrishnaIyer