TheThingsNetwork/lorawan-stack #111: Add Nixpkgs expression

rvolosatovs, 2019-02-07:
**Summary:** 

Package TTN stack and CLI for https://github.com/NixOS/nixpkgs
Adding a NixOS module for ease of deployment could be a nice next step.
References #86 

**Why do we need this?**

To make users happy

**What is already there? What do you see now?**

Nothing

**What is missing? What do you want to see?**

Nixpkgs expression.

**How do you propose to implement this?**

Similar to https://github.com/NixOS/nixpkgs/blob/d499b7d821106964581cbb6eab5354fbdf7a6847/pkgs/development/tools/richgo/default.nix#L1-L21
We just need to set `TTN_LW_HTTP_STATIC_SEARCH_PATH` for stack in a wrapper, same as it's done for Brew in #109 
We can use https://github.com/adisbladis/vgo2nix to generate most of the expression.(Note, however, that I had an issue with Thrift dependency when I tried to do that)

htdvisser, 2021-03-26:
Is there any demand for this? Otherwise I think we should just close this.

rvolosatovs, 2021-03-26:
This is something that needs to be done before the demand is there.

There's a big difference between a user just doing
```sh
nix run nixpkgs.lorawan-stack -c ttn-lw-stack start
```
and having stack running in their terminal window immediately with no setup and them going here and filing an issue, setting up Go and whatnot.

rvolosatovs, 2021-03-26:
Additionally, if we want to encourage open-source developers to actually develop, I think we should reconsider adding Nix support to out project via `shell.nix`. @johanstokking I open a PR before, but you rejected it (https://github.com/TheThingsIndustries/lorawan-stack/pull/256)

johanstokking, 2021-03-26:
> and having stack running in their terminal window immediately and them going here and filing an issue, setting up Go and whatnot.

This would be useful for the CLI. The Stack isn't something you usually start from the terminal as a non-developer.

> @johanstokking I open a PR before, but you rejected it

That was when we were just a few months in development with TTS, 1,5 year before our first release, so at the time it didn't make sense to spend time on that.

htdvisser, 2021-03-30:
As just discussed, we'll keep the issue open, but don't work on it unless it gets some upvotes.

KrishnaIyer, 2025-04-30:
This is not something we are likely to support.

closed 2025-04-30 by KrishnaIyer