TheThingsNetwork/lorawan-stack #112: Label Docker images

rvolosatovs, 2019-02-07:
**Summary:** 

Docker images should be labeled. See http://label-schema.org/rc1/

**Why do we need this?**

Because it's good practice

**What is already there? What do you see now?**

No labels

**What is missing? What do you want to see?**

Labeled Docker images

**How do you propose to implement this?**

By passing the labels as arguments in `build_flag_templates` in `docker` section of `.goreleaser.yml` introduced in #109. See https://goreleaser.com/customization/#Docker for more info.

**Environment:**

CI

htdvisser, 2021-03-26:
Is there any demand for this? Otherwise I think we should just close this.

htdvisser, 2021-03-26:
Actually, this is really small, I'll see if I can do this today.

closed 2021-03-30 by htdvisser