TheThingsNetwork/lorawan-stack #113: Add OpenWRT package

rvolosatovs, 2019-02-07:
**Summary:** 

Package TTN stack and CLI for OpenWRT.
References #86 

**Why do we need this?**

To make users happy

**What is already there? What do you see now?**

Binary releases

**What is missing? What do you want to see?**

OpenWRT package.

**How do you propose to implement this?**

Looks like `goreleaser` does not have support for it, but we can maybe add it?
Brew tap in https://github.com/TheThingsNetwork/homebrew-lorawan-stack should serve as an example.

**Environment:**

CI

htdvisser, 2021-03-26:
Is there any demand for this? Otherwise I think we should just close this.

rvolosatovs, 2021-03-26:
Same reasoning as in all the other ones in this series.

If I have a device using OpenWRT and I get to choose a LoRaWAN stack to put there, I am going to choose the one already available in the package repository.

htdvisser, 2021-03-30:
As just discussed, we'll keep the issue open, but don't work on it unless it gets some upvotes.

vigalent, 2022-04-25:
I think some users may want to install the stack on their gateway in order to create a on premise / private service and pass data to other services.

meckiemac, 2022-04-25:
> I think some users may want to install the stack on their gateway in order to create a on premise / private service and pass data to other services.

Yes right and I have an example: What I currently do is copy manually the temporary keys (NwkSkey & AppSKey) out of TTN and put them into the ABP encryption to use the data also locally forwarded to an MQTT Server. Would be nice to automate this locally on the gateway.