TheThingsNetwork/lorawan-stack #120: Always use GS downlink scheduler

johanstokking, 2019-02-09:
**Summary:** 

Always use GS downlink scheduler; do not use the gateway's if supported.

**Why do we need this?**

1. To enforce duty-cycle in GS
2. To avoid scheduling conflicts in GS without doing a gateway roundtrip
3. To report delay to NS in order to schedule a bunch of class C downlink over time (#78)
4. To allow optimizing downlink paths, i.e. try RX1s of multiple gateways, then try their RX2

**What is already there? What do you see now?**

GS allows frontends to have scheduling done by the gateway. This is built for Basic Station which contains a scheduler. We're probably making use of this in #74 (cc @KrishnaIyer).

**What is missing? What do you want to see?**

GS should always use its own scheduler. This means that connection setup becomes simpler as we can get rid of some conditions.

**How do you propose to implement this?**

Always instantiate scheduler in GS, remove conditions and arguments that allow for a gateway scheduler.

**What can you do yourself and what do you need help with?**

Support @KrishnaIyer

johanstokking, 2019-03-25:
For the record, the reasons why we don't use the Basic Station internal scheduler is;

1. It doesn't track duty-cycle, so we can't safely keep sending RX2 options which, if chosen, violate duty-cycle
2. It doesn't send any message back to LNS whether scheduling worked out or which RX window has been chosen
3. The class C downlink is not deterministic; the Basic Station chooses a "convenient time", but that means we can't do conflict detection. And as conflicts are not reported, this mechanism is unreliable fire-and-forget which is not good enough

Therefore, we should use only "class A" mode downlink, also for class C. For any absolute time bound scheduling (class B or C), we may use "class B" mode downlink, or also class A as GS does time synchronization (per #180).

This means that the Basic Station frontend (#74) should figure out an artificial `xtime` that matches a reference time in the past. The frontend needs to know if RX1 or RX2 was chosen to calculate the right `xtime`. The RX window is currently not part of the `TxSettings`, so I'm suggesting to introduce a `TxSettings.Scheduled` with RX window number, and while we're at it, downlink class.

KrishnaIyer, 2019-03-25:
https://github.com/TheThingsNetwork/lorawan-stack/issues/358

I think that this issue along can be closed once the linked PR (#330 ) is merged.

PR #330 (2019-03-25): Enforce use of GS Scheduler for all gateways
**Summary:**
Closes #120.

**Changes:**

- Remove all functions and conditions that support using the Gateway Scheduler.

**Notes for Reviewers:**
Nil

**Release Notes: _(optional)_**

Nil
closed 2019-03-25 by KrishnaIyer