TheThingsNetwork/lorawan-stack #124: Expand Getting started document

mrpackethead, 2019-02-11:
**Summary:** 
Expand the getting started a document with more detail on how to install the stack, and use the CLI, particularly using the CLI remotely. 

**Why do we need this?**
Without this information, its very difficult for someone who wants to stand up the stack ( who is not a stack developer ) for testing their devices/gateways against is very difficult. Its a little bit of a black hole. 

**What is already there? What do you see now?**
The getting started a document in the git repo

**What is missing? What do you want to see?**

(1) Expanding the documentation with more detail to install 
(2) Expanding information on how to use the CLI both locally and remotely. 
...

**How do you propose to implement this?**
If someone could walk me through the process, I would be happy to write 'sys-admin' level documents for people to install on *nix type systems. 

**What can you do yourself and what do you need help with?**
I can write / edit / expand on current documents.   I need some help with being shown how to do it once.   I have a system available running Debian 9, in AWS, which can be used.

johanstokking, 2019-02-11:
Also adding class C example to the wish list.

htdvisser, 2019-02-13:
As soon as a v3.0.0 release is tagged, we should stop recommending using the cli in Docker, but use the locally installed (with apt/brew/binary...) version instead.

johanstokking, 2019-02-26:
I'm doing a quick clarification/typo roundup

PR #179 (2019-02-26): Getting started improvements
**Summary:**

Closes #156 
Closes #124

**Changes:**

- Add the changes that were requested before the open-source release (mostly formatting).
- The CLI `--help` flag is now suggested for more information regarding the CLI itself.
- The guide no longer instructs users to use the CLI from inside docker.
- Self-signed certificates now bear a more clear warning that browsers will reject them.
- Invalid `ENV` variables in the `docker-compose.yml` file have been fixed

**Notes for Reviewers:**

This should be merged only after the tagged release, since it relies on the [... cut]
PR #201 (2019-02-27): Minor getting started typo fixes and clarifications
**Summary:**

Closes #124 

**Changes:**

- Fix typos
- Clarify a bit
- Congratulate the user
closed 2019-02-27 by johanstokking