TheThingsNetwork/lorawan-stack #126: Console global notifications

bafonins, 2019-02-11:
**Summary:** 

We need a notification system. Can be similar as it is in the v2 console. 

**Why do we need this?**

We need this to inform the user about successful/failed/pending(?) operations, e.g. creating/editing/etc. entities. It can be implemented as a part of the form, however, some operations require immediate redirects, like the deletion operations, it is less obvious how to implement such notification in these cases. 

**What is already there? What do you see now?**

* The notification component
* Form `handleSubmit` callbacks

**What is missing? What do you want to see?**

* Specific reducer for global notifications
* Floating container for notifications
* Success color in the console styleguide

**How do you propose to implement this?**

Use 3rd party library(?)

**Environment:**

Browser

**What can you do yourself and what do you need help with?**

Implement it

@kschiffer

kschiffer, 2019-02-11:
Yup. We should indeed try and look for a library here. The only one that I could find that is not completely overbloated and allows for custom styling is this one: https://github.com/BananaBobby/react-awesome-toasts

We should use notifications for these scenarios:
- When clarity of the feedback of a user action would otherwise be insufficient
  - This is usually when updating (using a form, not in-place) or deleting entities
  - When entities are created, a simple redirect to the newly created entity should, however, be sufficient
- When *important* events (usually errors) that did not directly originate from a user action need to be brought to his/her attention

bafonins, 2019-02-12:
>  When _important_ events (usually errors) that did not directly originate from a user action need to be brought to his/her attention

Does this include the connection status? Do we want to inform the user that the data in [data streams](https://github.com/TheThingsNetwork/lorawan-stack/issues/28) cannot be updated via global notifications or via the `DataPanel` component?

What would be a proper position for the notifications on the screen?
Do we want to stack them as it is implemented in v2 console? If so, do we want to show the latest notification on the top/bottom?

bafonins, 2019-02-12:
Also, would it be a good idea to extend the `Notification` component with the title?

kschiffer, 2019-02-12:
> Does this include the connection status? Do we want to inform the user that the data in [data streams](https://github.com/TheThingsNetwork/lorawan-stack/issues/28) cannot be updated via global notifications or via the `DataPanel` component?

I believe its best to use local scopes whenever possible. A connection error in the data stream is directly connected to the `DataPanel` component, hence the component should provide a means to display these issues.

> What would be a proper position for the notifications on the screen?

Bottom or top right, I'd slightly lean towards bottom. If we use top, it should be below the navbar.

> Do we want to stack them as it is implemented in v2 console?

I don't like stacking notifications. If that occurs, it is an indication that we are issuing too many notifications (likely with the same content). If we have to issue multiple notifications, they should run in sequence **and** preferably check whether their contents are not the same.

>Also, would it be a good idea to extend the `Notification` component with the title?

Yes.
[image]

Also, for clarity, we should actually call what we are discussing now "toasts" which seems to be the correct term. Notifications is a little ambiguous. We will likely also have a notification system in the future (to be understood as persistent system messages).

kschiffer, 2019-02-12:
Some more considerations when using toasts:
- Should be used scarcely
- Wording should be short and concise
- They should dismiss themselves by default
  - Duration should be no shorter than 4sec and at best based on content length
  - Hovering over them should pause the timer
- We could consider providing interactive elements (like a "retry" link) where applicable

closed 2019-02-20 by bafonins