TheThingsNetwork/lorawan-stack #136: Allow users to request being added as collaborator

htdvisser, 2019-02-13:
**Summary:** 

It would be nice if we could allow users to request having themselves added as collaborator on an entity.

**Why do we need this?**

Users may want to gain access to organizations, applications, gateways etc. without the existing collaborators knowing, and without a way for the user to contact an existing collaborator.

**What is already there? What do you see now?**

Users can see public information of an entity. For OAuth clients they can also see the list of collaborators (so that they know who may gain access to their data when they authorize the client). Usually however, users will want to join other entities.

**What is missing? What do you want to see?**

A way for users to indicate that they want to collaborate on an entity.

**How do you propose to implement this?**

We could extend the membership model with a `state`, similar to how users and OAuth clients have a state. If "membership requests" are enabled for an entity, users can create memberships on the entity, but only in the "requested" state, which will not give the user any of the rights that the membership model contains. This will trigger notification to existing collaborators (or the contact info) of the entity.

Existing collaborators with rights to manage the entity will be able to see memberships in the "requested" state, and edit these memberships in order to make them "accepted".

This requires a database migration for the IS, as we'll need to convert existing memberships to the "accepted" state.

This may require modifications to the interface of `MembershipStore`'s `SetMember`.

**What can you do yourself and what do you need help with?**

I can implement this myself, or hand it over to another Identity Server expert

htdvisser, 2022-07-07:
Removing assignment here. If this is something users want, please react with a 👍 vote on the issue.

KrishnaIyer, 2025-04-30:
This isn't a very common use case and is not requested for the years since this was created. I think it's safe to close this one.

closed 2025-04-30 by KrishnaIyer