TheThingsNetwork/lorawan-stack #137: Support TLS client auth in CUPS

htdvisser, 2019-02-13:
**Summary:** 

We should support TLS client auth in our CUPS server.

**Why do we need this?**

Because it is one of the valid auth methods for Basic Station gateways.

**What is already there? What do you see now?**

When #75 is done we'll have a CUPS server with TLS Server auth. The client authenticates with a token.

**What is missing? What do you want to see?**

We should also support TLS Client auth.

**How do you propose to implement this?**

We can look at `c.Request().TLS.PeerCertificates` if the CUPS server is facing the client directly. We'll need an alternative if we use a TLS-terminating proxy.

**What can you do yourself and what do you need help with?**

I'll first finish #75, and we can look into this when it's actually needed.

ogimenez-smtc, 2020-10-30:
Hello @htdvisser , any schedule for that feature ? Thank you

htdvisser, 2020-10-30:
No. Since token auth works just fine, there is not a lot of demand for it, so the core team is working on features with higher priority. Of course, our community is welcome to pick this up. Would you be interested in contributing this feature yourself?

ogimenez-smtc, 2020-10-30:
ok thank you for the status, I am too limited in Go knowledge to implement it :(

It is fine to use auth tokens as long as #3426 is fixed

KrishnaIyer, 2022-11-15:
This is covered by an enterprise PR internally.

closed 2022-11-15 by KrishnaIyer