TheThingsNetwork/lorawan-stack #143: Registration of Clusters in Identity Server Entity Registry

htdvisser, 2019-02-14:
**Summary:** 

We should add the new entity "Cluster" to the Identity Server's Entity Registry.

**Why do we need this?**

1. Discoverability. Right now there is no way get an overview of clusters one can connect to. With registration we could give users an overview of clusters they can choose from.
2. Authentication. Right now there is no way for (external) clusters to get private data from the IS Entity Registry about entities they are responsible for. Think about the Gateway Server needing private data of the gateways that are linked to it, or the NS/AS/JS needing private data of the devices that are registered on them.

**What is already there? What do you see now?**

We currently use "in-cluster auth" for this, which works fine for private networks.

**What is missing? What do you want to see?**

In-cluster auth works with a shared secret, so that won't work for a setup with a central IS and multiple clusters managed by different parties. If clusters would be registered entities, they could authenticate to the IS with API keys, and we would only need to verify that the {Gateway,Network,Application,Join}ServerAddress of the entity they're requesting matches their registered address.

**How do you propose to implement this?**

Add a "Cluster" entity, with an immutable address, a state like OAuth Clients (because clusters will need to be approved), possibly a public key (if clusters would need to talk to other clusters, we can't use the API keys, so we'll need to sign their requests). Alternatively, the IS could become a CA and sign certificates for clusters. The Cluster could be linked to an OAuth Client as each cluster will have a console as well. 

Where clusters talk to the IS, the in-cluster auth could be replaced by Cluster API Key. In places where in-cluster auth is currently used for communication to an (external) NS/AS/JS this would have to be replaced by a method that doesn't leak the API key nor in-cluster shared secret.

**What can you do yourself and what do you need help with?**

I can take care of the API, the IS ER part and create `grpc.CallOption`s that inject the Cluster API Key into requests to the IS (I think we can determine if this is the case from the `uri` arg in `GetRequestMetadata`). We'll need to discuss communication between NS/AS/JS and possible request signatures or TLS Client Auth. @johanstokking

johanstokking, 2019-02-18:
I think we need to take care of naming here. We currently have "cluster authentication" with a "cluster key" and introducing a "cluster API key" gets confusing. I think on this level, we're talking about "network authentication".

I also think we need that we need TLS client authentication for clusters to authenticate to others; using a NS or JS in another cluster is pretty realistic.

The cluster with the IS then uses a self-signed certificate to dial to JS/NS/AS I presume?

If we have TLS client authentication, what do we have the cluster API key for?

>In places where in-cluster auth is currently used for communication to an (external) NS/AS/JS this would have to be replaced by a method that doesn't leak the API key nor in-cluster shared secret.

Which places are these? Because this is definitely wrong if this happens. This doesn't happen in NS-AS and afaik not in NS-JS nor AS-JS either.

htdvisser, 2019-02-18:
The term "Cluster API Key" is indeed a bit unfortunate, but it logically follows from having API keys on the Cluster entity, similar to how User API Keys are API keys on the User entity.

I would be against using "Network" instead of "Cluster". The "Network" is the entirety of "Clusters", so if we're looking for a different name to avoid confusion, perhaps we could use "Region".

If we're going with TLS Client Auth, then the Identity Server becomes a CA with its own Root certificate. That Root certificate will need to be trusted by other clusters, and would be the parent of the certificates that the IS signs for registered Clusters. Bootstrapping a cluster that contains the IS might become a bit tricky, but outgoing calls would work exactly the same.

The receiving side of the request (let's say the JS) has to verify that the TLS client credentials (which contain the cluster address) match the NetworkServerAddress that is registered in the stored EndDevice model.

I was not aware of any places where cluster auth is wrongly used, but we should check it nonetheless.

johanstokking, 2019-02-18:
>I would be against using "Network" instead of "Cluster". The "Network" is the entirety of "Clusters", so if we're looking for a different name to avoid confusion, perhaps we could use "Region".

Because of the entirety it's that level of authentication. But network is confusing too, region as well – single clusters are networks too.

I agree that cluster API key would be consistent with the naming we have for others. If we go with cluster API keys, we can better rename the cluster key. But with TLS client authentication I don't see a need for cluster API keys in the first place. For cluster management we can use TLS client authentication as well.

htdvisser, 2019-06-25:
Additional field to register with each cluster: a list of JoinEUI prefixes (#138)

@johanstokking I don't think we have more naming options than "Cluster" or "Region"

johanstokking, 2019-06-25:
Yeah let's go with cluster API key then. I have no better suggestions either and it's consistent with other entity API keys. I'm not too afraid of confusion either as cluster auth keys are not something that 99% of the users will work with.

htdvisser, 2019-07-09:
Blocking #138

kschiffer, 2020-03-21:
Do we have a timeline here? This is crucial to improve the UX during onboarding since users get stuck on the OAuth landing view.
See also https://github.com/TheThingsIndustries/lorawan-stack/issues/1968

htdvisser, 2020-09-29:
- This is no longer blocking #138 (we went for DNS there)
- This is no longer a priority for UX (we went for a configurable cluster picker instead of a dynamic one)

htdvisser, 2022-07-20:
Removing my assignment and bumping back to triage for re-assignment and prioritization (cc: @NicolasMrad).

KrishnaIyer, 2023-07-04:
This can go back to the backlog until we look at it again.

KrishnaIyer, 2025-04-30:
> Discoverability. Right now there is no way get an overview of clusters one can connect to. With registration we could give users an overview of clusters they can choose from.

This is actually a valid requirement that we can pick up separately and we'll reconsider the proposed implementation later.

closed 2025-04-30 by KrishnaIyer