TheThingsNetwork/lorawan-stack #147: HTTP error: urls cannot be blank.

essamsalah, 2019-02-14:
**Summary:** 
Unable to login from CLI on a fresh Docker installation.

**Steps to Reproduce:** 

- Installing V3 stack using Docker as per [gettingstarted.md](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/doc/gettingstarted.md) 
- Reaching the login step `docker-compose exec stack ttn-lw-cli login`
- Copy the url into the browser, login using the correct username and password
- Got a blank page with error message saying `HTTP error: urls cannot be blank.`

**What do you see now?**
Blank page in the browser with error message

**What do you want to see instead?**
Authorization code

**Environment:**

- macOS 10.13.6
- node v10.11.0
- go version go1.11.5 darwin/amd64
- Yarn 1.13.0
- Docker 18.09.2
- docker-compose version 1.21.2, build a133471

**What can you do yourself and what do you need help with?**
As per slack conversation, it seems that there need to be a proper sync between documentation on Github and Docker images, so this seems a dead-end for now.

htdvisser, 2019-02-15:
Refs #109 #128. 

This should be fixed as soon as we have a new tagged release. From that moment on, we also should point to the documentation (readme, getting started, etc.) of that tagged release, and not to `master`.

Possible workarounds for now: 

1. Build the CLI from source (master branch) and use that instead.
2. Update the CLI's registration in the database with something like `update clients set redirect_uris = ARRAY['http://localhost:11885/oauth/callback'] where client_id = 'cli';`

essamsalah, 2019-02-18:
using option #2, the error is not reproduced any more and the Authorization code is presented. However, there are other issue as below image ...

[image]

htdvisser, 2019-02-18:
For the CLI to be able to do the exchange, it needs to trust the server. It looks like you only trusted the development certificate in your browser (otherwise you wouldn't have gotten the code), and not in your OS (otherwise the CLI would have trusted it too.

If you don't want to trust your development certificate in your OS you can manually add it to the CLI with the `--ca` flag or the `TTN_LW_CA` environment variable.

@adriansmares the Getting Started may need to document clearer that users need to actually trust the certificate that they generate in "Dependencies".

johanstokking, 2019-02-18:
Issue seems resolvable by the poster. We'll update getting started accordingly.

closed 2019-02-18 by johanstokking