TheThingsNetwork/lorawan-stack #150: make clean build-all fails

essamsalah, 2019-02-15:
**Summary:** 
Unable to complete the build script for the V3 development environment

**Steps to Reproduce:** 

- Following the steps from [Development Environment](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/DEVELOPMENT.md) 
- just after running `make test`
- run `make clean build-all`

**What do you see now?**
[Screenshot](https://imgur.com/a/H8W3gNA)

**Environment:**
- go version go1.11.5 linux/amd64
- node  v10.15.1
- yarn  1.13.0
-  Ubuntu /AWS T2.small EC2

essamsalah, 2019-02-15:
@adriansmares  advised the following steps before `make clean build-all`:

`
cd <repo-folder>
sudo rm -rf node_modules
chown -R <username>:<username> .
make dev-deps
make deps
`

johanstokking, 2019-02-18:
@essamsalah does that fix the issue?

How did you clone the repository in the first place, what were the folder rights?

bafonins, 2019-02-18:
see https://github.com/TheThingsNetwork/lorawan-stack/issues/157#issuecomment-464900724

closed 2019-02-19 by bafonins