TheThingsNetwork/lorawan-stack #157: build all fails on make js.build

pdubois22, 2019-02-18:
**Summary:** 
on a clean ubuntu install, make build-all or make js.build fails

**Steps to Reproduce:** 
clean ubuntu install

git clone https://github.com/TheThingsNetwork/lorawan-stack.git
cd lorawan-stack
make init
make deps

ubuntu@ubuntu:~/lorawan-stack$ make js.build
lorawan-stack  Building JS SDK files
yarn run v1.12.3
$ babel src -d dist --no-emoji --no-progress --quiet
Done in 1.17s.
lorawan-stack  pre-building config files [babel config/webpack.dll.js]
make js.webpack-dll
make[1]: Entering directory '/home/ubuntu/lorawan-stack'
lorawan-stack  building dll file
   535 modules                                                                              

ERROR in dll libs
Module not found: Error: Can't resolve 'ttn-lw' in '/home/ubuntu/lorawan-stack'
.make/js/build.make:80: recipe for target 'js.webpack-dll' failed
make[1]: *** [js.webpack-dll] Error 2
make[1]: Leaving directory '/home/ubuntu/lorawan-stack'
.make/js/build.make:77: recipe for target 'public/libs.bundle.js' failed
make: *** [public/libs.bundle.js] Error 2

**What do you see now?**

ERROR in ./pkg/webui/console/api/index.js

**What do you want to see instead?**
build as in the doc

**How do you propose to implement this?**
not sure about the :
import TTN from 'ttn-lw'

in pkg/webui/console/api/index.js 

**Environment:**
fails on fresh ubuntu, and MacOSX

**What can you do yourself and what do you need help with?**
try to figure out what is wrong with my setup

johanstokking, 2019-02-18:
This may be related to #150

bafonins, 2019-02-18:
The issue is in the order of the javascript build process. Currently, the v3 console uses the javascript sdk that is under development and imports the sdk as a local dependency. This means that the sdk should be built before the dependencies of the console are installed, which is not the case at the moment.

Quick solution would be to add the sdk manually via yarn:
`yarn add file:sdk/js`
if you dont have yarn installed you can fetch the js deps again with:
```
rm -rf node_modules
make js.dev-deps js.deps
```

I will revisit the javascript make targets tomorrow and fix the issue.

pdubois22, 2019-02-19:
yarn add file:sdk/js works as a workaround

closed 2019-02-19 by bafonins