TheThingsNetwork/lorawan-stack #159: segmentation violation ns on downlink to ABP device downlink.go l 167

pdubois22, 2019-02-19:
**Summary:** 
registered a device as an ABP device (--supports-join=false).
ns crashes on downlink

exit status 2

**Steps to Reproduce:** 
register a device on the ns that is already on the field:

go run ./cmd/ttn-lw-cli devices create test dev2 --defaults=true \
--supports-join=false \
--join-eui=70B3D59520000000 --dev-eui=A81758FFFE032277 \
--root-keys.root-key-id=dev2 \
--root-keys.app-key.key=4F939595CE83288AFEF81BD881C3C36E \
--session.dev-addr=010B600C \
--session.keys.app-s-key.key=A15BEF4A3233274AE917E4AFB19055F2 \
--session.keys.f-nwk-s-int-key.key=9E2FB61D7310C9279886DB79FA52F9F4 \
--lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
--frequency-plan-id EU_863_870 \
--mac-settings.use-adr=true \
--resets-f-cnt=false \
--resets-join-nonces=false \
--session.last-f-cnt-up=45872 \
--session.last-n-f-cnt-down=1880 \
--uses-32-bit-f-cnt=true

**What do you see now?**
ns crash on downlink.go l 167:

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x215c15c]

goroutine 219 [running]:
go.thethings.network/lorawan-stack/pkg/networkserver.generateDownlink(0x28f7a20, 0xc000eba600, 0xc0002dc6c0, 0xe60019, 0xc000168070, 0x28f7a20, 0xc000eba5a0, 0x3, 0x0, 0xc00023a560, ...)
	/Users/pdubois/lorawan-stack/pkg/networkserver/downlink.go:167 +0xafc

**What do you want to see instead?**
ns handling gacefully uplink and downlink.

**How do you propose to implement this?**
on line 167, EndDeviceIdentifiers is nil. 
i think the create command does not create the dev ids:

go run ./cmd/ttn-lw-cli devices get test dev2 --session
{
  "ids": {
    "device_id": "dev2",
    "application_ids": {
      "application_id": "test"
    }
  },
....

**Environment:**
macOS, but crashes also on Centos and Ubuntu.

**What can you do yourself and what do you need help with?**

rvolosatovs, 2019-02-19:
This is fixed in #101

pdubois22, 2019-02-20:
@rvolosatovs : no more panic, but cannot register ABP:

go run ./cmd/ttn-lw-cli devices create test dev3 --defaults=true \
> --supports-join=false \
> --join-eui=70B3D59520000000 --dev-eui=A81758FFFE031BF9 \
> --session.dev-addr=FC006A0C \
> --session.keys.app-s-key.key=2412B7B219FCEE78BDFACB5944764116 \
> --session.keys.f-nwk-s-int-key.key=57A976DF984AF3EC514B21E56C081510 \
> --lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
> --frequency-plan-id EU_863_870 \
> --session.last-f-cnt-up=25860 \
> --session.last-n-f-cnt-down=901
 ERROR Could not create end device, rolling back... error=error:pkg/networkserver/redis:invalid_identifiers (invalid identifiers)

not sure which identifier is wrong.
no message on master. can register
btw, if device is registered before applying #101 to master, i got a wrong wireType error. I guess we need to delete / recreate devices after #101 is applied, which is ok:

  WARN Failed to find devices in registry by DevAddr dev_addr=FC006A0C error=proto: wrong wireType = 0 for field RootKeys grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc payload_length=25 request_id=01D44XF213HJHY8VPZNPZEPYB1 uplink_f_cnt=26019

rvolosatovs, 2019-02-20:
Please try the latest changes on #101

pdubois22, 2019-02-20:
fixed. thanks.

if you want me to do some extra testing after further refactoring of #101, let me know.

PR #101 (2019-02-22): Overriding desired MAC parameters and defaults
**Summary:**

References #5 
Closes #60 
Closes #104 
Closes #105 
Closes #144 
Closes #159 

**Changes:**

- Add all optional fields from Backend Interfaces `DeviceProfile` as well as additional MAC parameters to `MACSettings`
- Ensure device API is aligned with `DeviceProfile ` from Backend Interfaces
- Make `MACSettings` optional
- Allow specifying NS-wide default `MACSettings` to fallback to if not defined by the device
- Update protoc image version (requires https://github.com/TheThingsIndustries/protoc-gen-fieldmask/pull/20)
- Set `RxDelay`  [... cut]
closed 2019-02-22 by rvolosatovs