TheThingsNetwork/lorawan-stack #163: Broken devices creation in AsDeviceRegistry via HTTP

kschiffer, 2019-02-19:
**Summary:** 
When attempting to create a device using `POST /as/applications/test-app/devices` (assuming that `test-app` exists), using the following (examplatory) payload:
```json
{  
   "end_device":{  
      "ids":{  
         "device_id":"device01",
         "application_ids":{  
            "application_id":"test-app"
         },
         "dev_eui":"0004A30B001C0543",
         "join_eui":"8000000000000019",
         "dev_addr":"26012531"
      },
      "application_server_address":"http://localhost:8888/test",
      "session":{  
         "last_a_f_cnt_down":121
      }
   },
   "update_mask":{  
      "paths":[  
         "ids",
         "session.last_a_f_cnt_down"
      ]
   }
}

```
yields:
```json
{
    "code": 3,
    "message": "error:unknown:unknown (invalid field Device.EndDeviceIdentifiers.DeviceID: value '' must be a string conforming to regex \"^[a-z0-9](?:[-]?[a-z0-9]){2,}$\")",
    "details": [
        {
            "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "message_format": "invalid field Device.EndDeviceIdentifiers.DeviceID: value '' must be a string conforming to regex \"^[a-z0-9](?:[-]?[a-z0-9]){2,}$\"",
            "attributes": {},
            "correlation_id": "83f2d2791ee94313873f45aa2d2ddadb"
        }
    ]
}
```

**Steps to Reproduce:** 
1. Create an application (with id: `test_app`)
2. Create a device in the AS registry using `POST /as/applications/test-app/devices` using the payload above

I'm assuming that the same issue will be true for the registries in JS and NS. However, creating a device in the IS worked fine for me.

**What do you see now?**
The error message above.

**What do you want to see instead?**
A message returning the newly created device.

**How do you propose to implement this?**
As suggested by @htdvisser, it might be that the grpc gateway matches the POST request to the PUT endpoint: `/as/applications/{device.ids.application_ids.application_id}/devices/{device.ids.device_id}` this will issue the error because the device id is not set.

**What can you do yourself and what do you need help with?**
I need somebody familiar with the grpc gateway to check for any issues or whatever else might cause this.

*Setting prio/high, as this blocks me from progressing device crud in the JS SDK (#131).*

htdvisser, 2019-02-19:
I don't think this is easy to fix, so please continue development with the primary HTTP mapping for `AsEndDeviceRegistry.Set`, which is `PUT /as/applications/{device.ids.application_ids.application_id}/devices/{device.ids.device_id}`

kschiffer, 2019-02-19:
Closing this due to false alert: happening because I was using the wrong field name, which is not set consistently. See #170.

closed 2019-02-19 by kschiffer