TheThingsNetwork/lorawan-stack #172: Delete api-keys from the cli

Oliv4945, 2019-02-20:
**Summary:** 
In the CLI we can create, list, update api-keys but there is no option to delete them

**Why do we need this?**
Revoke an api-key if it gets disclosed, unused

**What is already there? What do you see now?**
`ttn-lw-cli app api-keys --help` returns
```
Available Commands:
  create      Create an application API key
  list        List application API keys
  update      Update an application API key
```

**What is missing? What do you want to see?**
delete or remove

**How do you propose to implement this?**
I propose to ask @htdvisser :)

htdvisser, 2019-02-20:
API keys (and collaborators) can be removed by updating them with an empty list of rights. I guess that it would make sense to add CLI commands for this for consistency with the other commands.

Oliv4945, 2019-02-20:
From user point of view, at least "delete" command wrapper would be nice :)

PR #193 (2019-02-28): CLI API-Key and collaborator deletion
**Summary:**
Closes #172

**Changes:**

- Added the `delete` command for `api-keys`
- Added the `delete` command for `collaborators`
- The set command for both API-keys and the collaborators no longer deletes the entity if no rights are set

**Notes for Reviewers:**

As discussed, set no longer deletes the entity since it would be counter-intuitive, given that the delete command exists separately.
closed 2019-02-28 by adriansmares