TheThingsNetwork/lorawan-stack #175: Enforce EUI and ID uniqueness in device registries

rvolosatovs, 2019-02-20:
**Summary:** 

Component registries should enforce uniqueness of IDs and EUIs.
This is already done in Network Server, but not in JS.

**Why do we need this?**

To avoid clashes

**What is already there? What do you see now?**

EUI uniqueness enforcement in JS, but not ID uniqueness

**What is missing? What do you want to see?**

Enforce uniqueness of ID and EUI pairs in registries

**How do you propose to implement this?**

Adapt from NS(https://github.com/TheThingsNetwork/lorawan-stack/blob/f1377b316a85ae3709a84db5b5f9a46faafba3a0/pkg/networkserver/redis/registry.go#L218-L233)
@johanstokking is AS relevant here too? If not, please remove the label.

johanstokking, 2019-02-21:
@rvolosatovs let's leave AS out of scope; the AS doesn't really care about EUIs.

johanstokking, 2019-02-21:
Let's do this for AS as well. Even if it was just as sanity check; it's an EUI.

PR #207 (2019-03-02): ID uniqueness, harmonization and namespacing in Redis registries
**Summary:**

Closes #175 
I noticed that registry implementations diverged quite a bit, included a bunch of redundant actions, weren't clear and had possible memory leaks.

**Changes:**

- Harmonized Redis registry implementations(not the signatures) - essentially I cleaned up NS registry implementation and then used that as the base for all other Redis registry implementations
- AS tests were throwing `panic`s on failing testcases, gave a bunch of useless output after a test failure, and did not handle registry errors, which I  [... cut]
closed 2019-03-02 by rvolosatovs