TheThingsNetwork/lorawan-stack #178: Collect "Release Notes" section of PRs in Goreleaser

htdvisser, 2019-02-21:
**Summary:** 

Instead of having Goreleaser generate a changelog from only PR titles, it should also look at the **Release Notes** sections of our pull requests.

**Why do we need this?**

The **Release Notes** sections may contain important notes about API changes, DB changes, config flag changes, etc.

**What is already there? What do you see now?**

- Goreleaser generates release notes (default)
- We keep a **Release Notes** section in our Pull Requests
- Goreleaser can take release notes from external generaters such as [this one](https://github.com/buchanae/github-release-notes) that they link to

**What is missing? What do you want to see?**

The release notes of a tag should contain the release notes sections that were written in the merged pull requests.

**How do you propose to implement this?**

Use the Github API to get the merged PRs, and use the (markdown) body (or the title to generate release notes). We may have to update the pull request template to specify the exact format that we want. Perhaps something like:

```
- NEW: is: describe a new feature in the identity server
- FIX: ns: describe a bugfix in the network server
- COMPAT: describe non-backwards-compatible changes
```

Perhaps we need some more discussion about what exactly we expect in the release notes, how much we want to automate (because we'll likely have to edit the release notes anyway), etc.

rvolosatovs, 2019-02-21:
Goreleaser uses commits, not PR titles(see https://github.com/rvolosatovs/lorawan-stack/releases/tag/v3.0.2 for example)

htdvisser, 2019-02-21:
Oh wow that's even worse. Let's not do that to our users.

johanstokking, 2019-02-21:
I agree. Those commits are easily accessible through GitHub anyway.

Hopefully we have an easy way to get the pull requests since the previous release.

PR #575 (2019-04-30): Generate release notes

### Summary:

Closes #178 

### Changes:

- Use headings in PR/issue templates
- Call https://github.com/TheThingsIndustries/release-notes to generate the release notes
closed 2019-04-30 by rvolosatovs