TheThingsNetwork/lorawan-stack #180: Convert downlink absolute time to timestamp if there's no GPS

johanstokking, 2019-02-21:
**Summary:** 

Convert downlink absolute time to timestamp if there's no GPS.

**Why do we need this?**

When scheduling an application downlink message to AS, users can specify an absolute time. This only works when the gateway has GPS sync. Downlink messages are failed by the gateway when there is no GPS lock. Falling back to a server-calculated timestamp would resolve this.

**What is already there? What do you see now?**

Converting the absolute time to GPS time and passing that as transmission time to the gateway.

**What is missing? What do you want to see?**

The current behavior but with fallback to a server-calculated timestamp.

> Note: gateways using "fake GPS" will be treated as GPS enabled gateways. That's the consequence of using this "feature".

**How do you propose to implement this?**

We should keep the existing implementation when the gateway indicates that it has GPS lock. This should be derived from the status it sends; i.e. presence of GPS coordinates.

When there is no GPS lock, the scheduler clock should be used to convert the absolute time to timestamp. This is already implemented in the scheduler.

**What can you do yourself and what do you need help with?**

I'll take this.

johanstokking, 2019-04-09:
It turns out that gateways need a GPS input for providing GPS timestamps, i.e. in UDP the `tmms` field, and the counterparts in other protocols.

This means that we cannot sync the gateway local time as the given time only has second precision which is not good enough.

Therefore, we go with the approach of measuring the roundtrip time from the server through the `TX_ACK` in UDP, and the counterparts in other protocols, if the protocol doesn't provide roundtrip already.

htdvisser, 2019-04-18:
Refs #398

FlorianLudwig, 2019-04-18:
It would be useful to have some accuracy measurement available to the application level.  

This should include a flag if gps of the gateway is used or a time synchronization like described by @johanstokking. In the case of the later it might also be helpful to have some data on the clock of the GS server but that might be out of scope for TTN.

johanstokking, 2019-04-23:
@FlorianLudwig we can do two things;

1. Provide the RTT as part of the [`GatewayConnectionStats`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/api.md#ttn.lorawan.v3.GatewayConnectionStats)
2. Have a flag in [`ApplicationDownlink.ClassBC`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/api.md#ttn.lorawan.v3.ApplicationDownlink.ClassBC) indicating the;
   - Requirement for GPS. If no GPS, fail the downlink (through `downlink_failed` in [`ApplicationUp`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/api.md#ttn.lorawan.v3.ApplicationUp))
   - If no GPS is permitted, specify a maximum allowed RTT value within which server scheduling is permitted. I.e., fail the downlink above 30 ms, let GS schedule blow 30 ms

What do you think?

FlorianLudwig, 2019-04-25:
@johanstokking 

> Provide the RTT as part of the GatewayConnectionStats

That would mean there is some aggregation already happening to RTTs.  I would prefer a way to get "raw" RTTs.

> Have some specifiers in ApplicationDownlink.ClassBC [...]
> Requirement for GPS. If no GPS, fail the downlink (through downlink_failed in ApplicationUp)
Agree, that sounds good.

> If no GPS is permitted, specify a maximum allowed RTT value within which server scheduling is permitted. I.e., fail the downlink above 30 ms, let GS schedule blow 30 ms

Which time would than be specified?  `gw_time = gs_time + avg(rtt)/2` ?

Having a maximum "uncertainty" range is a great idea.

## specifying xtime / gateway time
It might also be useful to specify the local gateway time instead of gps time when scheduling a message from the application.

## Uplink messages
Reading the specifications it seems that the basic station protocol also does not contain a wall clock time for uplink messages from a gateway without gps, is that correct?

## A complete different option
Might it be feasible to change the implementation of the gateway software to include the local time even with no GPS available?

johanstokking, 2019-04-25:
>That would mean there is some aggregation already happening to RTTs. I would prefer a way to get "raw" RTTs.

I see. Let's make this available through Prometheus then.

>Which time would than be specified? `gw_time = gs_time + avg(rtt)/2` ?

The median, but indeed.

>It might also be useful to specify the local gateway time instead of gps time when scheduling a message from the application.

The problem is that we don't know it. We only get it with second precision, which is far from good enough.

>Reading the specifications it seems that the basic station protocol also does not contain a wall clock time for uplink messages from a gateway without gps, is that correct?

It does, in fact, see [Monitoring Round-trip Times](https://doc.sm.tc/station/tcproto.html#lorawan-data-frames). An example is `RefTime: 1546596145.625418`, which contains enough second fraction precision to be used. This allows for syncing `xtime` with the gateway local time. If the gateway local time is good (through NTP), we can use that, while not depending on GPS.

>Might it be feasible to change the implementation of the gateway software to include the local time even with no GPS available?

In UDP, we already have second precision. We can change this to sub-second precision. But I'm not a huge fan of modifying UDP forwarder as that's a rabbit hole with no return. In Basic Station, this is set through the round-trip time measurement that's part of the protocol. In UDP, we do the same, but then through `TX_ACK`, and without depending on the gateway local time.

htdvisser, 2019-04-25:
> Let's make this available through Prometheus then.

I don't think it's smart to put these kind of gateway connection stats in Prometheus. These will generate *a lot* of series (the number of gateways times the number of GS instances) with millions of data points (depending on the screpe interval). Do we really need to see the evolution of the RTT for each gateway over time? Isn't it sufficient to have the `n` most recent measurements in the `GatewayConnectionStats`?

johanstokking, 2019-04-25:
@FlorianLudwig can you elaborate on your requirements w.r.t. the RTT over time? Would a sliding window with mean/min/max/count be informative enough?

PR #589 (2019-05-01): Absolute time scheduling with round-trip times

#### Summary

Closes #180 
Closes #398 

#### Changes

- Introduce generic 16-bit downlink token that's aware of time
- Implement and integrate round-trip time recording in UDP frontend
- Make RTT statistics available through GS connection stats
- If RTT available, use maximum value for "late" detection
- If RTT available and no gateway time available, use median value for absolute time scheduling

#### Notes for Reviewers

@rvolosatovs only review API as a formality

#### Release Notes

- Added round-trip times to Gateway Server connection stati [... cut]
closed 2019-05-01 by johanstokking