TheThingsNetwork/lorawan-stack #181: Harmonize device registry get/set signatures

johanstokking, 2019-02-21:
**Summary:** 

Harmonize device registry get/set signatures.

**Why do we need this?**

For a consistent developer experience.

**What is already there? What do you see now?**

1. AS device registry takes `ttnpb.EndDeviceIdentifiers` and its primary identifiers
2. NS device registry takes the primary identifiers as strings

The reason I don't like 2 is that everywhere in the codebase we assume that `ttnpb.*Identifiers` has filled out primary identifiers. Otherwise validators fail. Otherwise the unique package panics. There is no reason to clarify that some method or function requires primary identifiers to be present as part of the call signature. `pkg/unique` doesn't even do this.

**What is missing? What do you want to see?**

NS to use same signature as AS.

**How do you propose to implement this?**

Copy paste interface methods and implement accordingly.

**What can you do yourself and what do you need help with?**

Review.

johanstokking, 2019-04-04:
Pushing this to Backlog.

This is a candidate to close; I don't think we should do this in the end after all the refactorings and harmonizations that we did in the device registries.

nejraselimovic, 2021-06-07:
@johanstokking should we close this or not?

johanstokking, 2021-06-10:
Let's leave open for future reference.

johanstokking, 2021-12-30:
This is already addressed. As far as there are any inconsistencies, we should address that as part of https://github.com/TheThingsIndustries/lorawan-stack/issues/2256

closed 2021-12-30 by johanstokking