TheThingsNetwork/lorawan-stack #187: Unable to make mac commands DeviceTimeReq/Ans work

Oliv4945, 2019-02-25:
**Summary:** 
I am unable make DeviceTimeReq/DeviceTimeAns work, the stack log shows an error:
```
DEBUG Deduplicating uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D4HW8WCVT4CR5JKRMZ04VX1H
stack_1      |  DEBUG Handling data uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D4HW8WCVT4CR5JKRMZ04VX1H
stack_1      |  DEBUG Matching device...                       dev_addr=00E4C4A3 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D4HW8WCVT4CR5JKRMZ04VX1H
stack_1      |  DEBUG Performing MIC checks on devices with matching frame counters... dev_addr=00E4C4A3 device_count=1 device_uid=app1.mote6 full_f_cnt_up=3 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs last_f_cnt_up=0 namespace=grpc payload_length=24 request_id=01D4HW8WCVT4CR5JKRMZ04VX1H uplink_f_cnt=3 uplink_f_cnt_up=3
stack_1      |  DEBUG Matched device                           dev_addr=00E4C4A3 device_uid=app1.mote6 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D4HW8WCVT4CR5JKRMZ04VX1H
stack_1      |  DEBUG Waiting for collection window to be closed... grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D4HW8WCVT4CR5JKRMZ04VX1H
stack_1      |  DEBUG Collection window closed, stopped deduplication grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D4HW8WCVT4CR5JKRMZ04VX1H
stack_1      |  ERROR Finished unary call                      duration=1.205224718s error=error:pkg/rpcserver:rpc_recovered (Internal Server Error) error_correlation_id=f4dc4a33f80246fabb108aa85f7be125 error_name=rpc_recovered error_namespace=pkg/rpcserver grpc_code=Internal grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D4HW8WCVT4CR5JKRMZ04VX1H
stack_1      |   WARN Finished unary call                      duration=1.206222037s error=rpc error: code = Internal desc = error:pkg/rpcserver:rpc_recovered (Internal Server Error) error_correlation_id=f4dc4a33f80246fabb108aa85f7be125 error_name=rpc_recovered error_namespace=pkg/rpcserver grpc_code=Internal grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc
stack_1      |  DEBUG Dropping message                         dev_addr=00E4C4A3 error=error:pkg/rpcserver:rpc_recovered (Internal Server Error) gateway_eui=7276FF0039030023 gateway_uid=ifemtobureau namespace=gatewayserver/io/udp
```

**Steps to Reproduce:** 
1. Send an uplink with DeviceTimeReq (0x0D) in the Fopts
2. Watch server logs for the failure

**What do you want to see instead?**
Stack sends a downlink with DeviceTimeAns to the device

**Environment:**
TTN stack is running under Docker, image 2098c207fc1d

htdvisser, 2019-02-26:
More details:

```
runtime error: index out of range
  File "/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/mac_device_time.go", line 50, in handleDeviceTimeReq
  File "/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/grpc_gsns.go", line 617, in func1
  File "/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/redis/registry.go", line 165, in func1
  File "/home/travis/gopath/src/go.thethings.network/lorawan-stack/vendor/github.com/go-redis/redis/tx.go", line 45, in Watch
  File "/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/redis/registry.go", line 143, in SetByID
  File "/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/grpc_gsns.go", line 442, in handleUplink
  File "/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/networkserver/grpc_gsns.go", line 973, in HandleUplink
```

closed 2019-02-28 by rvolosatovs

htdvisser, 2019-02-28:
@Oliv4945 the error is now resolved, but if you try again with the code in the current `master` and the same setup from before, you'll likely still not get an answer because it looks like was no GPS time in the uplink.

PR #208 (2019-02-28): Drop DeviceTimeReq if time is unknown
**Summary:**

Closes #187

**Changes:**

- Fix panic