TheThingsNetwork/lorawan-stack #189: Checkbox group component

bafonins, 2019-02-25:
**Summary:** 

Implement the checkbox group component

**Why do we need this?**

We need some convenient way to manage lists of checkboxes.

**What is already there? What do you see now?**

- The `Field` component
- The `Checkbox` component

**What is missing? What do you want to see?**

A solution that can be integrated into the console. Something like this:
[image]
Also see [User-Friendly Multi-select](http://doejo.com/blog/solutions-for-implementing-user-friendly-multi-select/)

The component must have:
1. Controllabel height and width
2. The 'select all` checkbox
3. Integration to our `Form` component
4. Display group related errors/validation, e.g. that at least on checkbox should be checked from the list

The component could have:
1. The search functionality
2. Additional list of current selections 

**How do you propose to implement this?**

tba

**Environment:**

Browser

**What can you do yourself and what do you need help with?**

Implement it

kschiffer, 2021-08-16:
Reminder for myself to include this in form updates.

KrishnaIyer, 2024-10-22:
We don't use such a component in any of our views. If we need one, we'll build it.