TheThingsNetwork/lorawan-stack #196: Dropped messages - uplink_channel_not_found on migrated devices

pdubois22, 2019-02-26:
**Summary:** 
Device that are migrated (devices that have joined on an other NS) and that are created on the stack using the cli (they have not joined on the stack), may send uplinks on channels that are not one of the 3 default channels on EU_863_870 (868100000, 868300000, 868500000).
Those uplinks on not default channels are dropped. 

**Steps to Reproduce:** 
When the device is created using the cli, the MACState Channels are initialized to the defaultChannels of the FP. (only 3 channels for  EU_863_870).
Uplinks received on  channels not declared in the MACState are dropped.

**What do you see now?**
 DEBUG Dropping message                         dev_addr=FC0068B1 error=error:pkg/networkserver:uplink_channel_not_found (uplink channel not found) gateway_eui=00000008004A5D13 gateway_uid=eui-00000008004a5d13 namespace=gatewayserver/io/udp
I have not seen a way to configure the channels on the cli.

Note that I think eventually, a MAC command will be send to adjust channels on the device, and therefore the MACState current channels will be extended to all possible values, and eventually, thinks may get ok, but that may take forever.

**What do you want to see instead?**
All EU Channels set in MACState upon creation in this case, to avoid message drops

**How do you propose to implement this?**
ability to change channels in cli. Having a custom FP is a workaround, but not ideal. 

**Environment:**
all

**What can you do yourself and what do you need help with?**
not sure what is the best solution. I wonder why MacState is initialised with only this subset of channels (Default channels).

johanstokking, 2019-02-26:
Thanks, you're just a few hours behind our own testing. We know the cause and will fix this in the next milestone.

johanstokking, 2019-03-04:
We have looked at this and it turns out that the LoRaWAN specification does not assume having the channel set up as part of the personalization procedure.

One could argue that the personalization encompasses what would otherwise be set up as part of the join procedure; i.e. device address, sessions keys, Rx delay and channels. However, the channels are not part of that; they need to be configured out-of-band. We have (`factory_preset_frequencies`)[https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/end_device.proto#L218] for that, which you can set as flag when creating and updating devices.

>All EU Channels set in MACState upon creation in this case

On a side note, there's no such thing as "all EU channels". EU868-like LoRaWAN networks are free to choose any channels in the ISM bands. This is different from US915-like plans, which does have channels, which, on contrary, will be all enabled.

closed 2019-03-04 by johanstokking

pdubois22, 2019-03-04:
You are a few hours beyond my own testing.
I got it working with mac_settings.factory_preset_frequencies

I agree with your comments.