TheThingsNetwork/lorawan-stack #197: API call with wrong credential (Bearer) return http code 404 instead of 401

pdubois22, 2019-02-26:
**Summary:** 
http API call for example to /api/v3/applications/xxx/devices/yyyy with an incorrect HTTP Authorization Bearer returns a 404.

**Steps to Reproduce:** 
execute an API call and mispell the bearer for example.

**What do you see now?**
404 return code

**What do you want to see instead?**
401 not authorized. 
404 should be reserved for device not found in that case.

**How do you propose to implement this?**

**Environment:**
API

**What can you do yourself and what do you need help with?**

htdvisser, 2019-02-26:
Thanks for reporthing this, @pdubois22. I'm unable to reproduce this with the information in this issue.

When I try misspelling "Bearer" as per your steps to reproduce, I get an error about the unsupported authorization method:

```
htdvisser % curl -H "Authorization: Bearero MFRWG.FKSG4PLAMFSTAUF7UWG45MGT4PRG3IBDV2NUVCA.OR6AK5JGXQLOMY7L6ASV3KINDMBUB4RW32OJUO2KUE6E6SLGZIKA" "localhost:1885/api/v3/applications/foo/devices/bar"
{"code":16,"message":"error:pkg/identityserver:unsupported_authorization (Unsupported authorization method)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/identityserver","name":"unsupported_authorization","message_format":"Unsupported authorization method","attributes":{},"correlation_id":"5292571136e84895a2b6d45e0cf7df9a"}]}
```

I also tried the normal flow. My user is an admin, so it is allowed to get device info:

```
htdvisser % curl -H "Authorization: Bearer MFRWG.FKSG4PLAMFSTAUF7UWG45MGT4PRG3IBDV2NUVCA.OR6AK5JGXQLOMY7L6ASV3KINDMBUB4RW32OJUO2KUE6E6SLGZIKA" "localhost:1885/api/v3/applications/foo/devices/bar"
{"code":5,"message":"error:pkg/identityserver/store:end_device_not_found (end device `foo:bar` not found)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/identityserver/store","name":"end_device_not_found","message_format":"end device `{application_id}:{device_id}` not found","attributes":{"application_id":"foo","device_id":"bar"},"correlation_id":"54cb551bf4904fc0915bef6910b5f5a9"}]}
```

And I tried with an incorrect key:

```
htdvisser % curl -H "Authorization: Bearer MFRWG.FKSG4PLAMFSTAUF7UWG45MGT4PRG3IBDV2NUVCA.OR6AK5JGXQLOMY7L6ASV3KINDMBUB4RW32OJUO2KUE6E6SL" "localhost:1885/api/v3/applications/foo/devices/bar"
{"code":7,"message":"error:pkg/auth/rights:no_application_rights (no rights for application `foo`)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/auth/rights","name":"no_application_rights","message_format":"no rights for application `{uid}`","attributes":{"uid":"foo"},"correlation_id":"daa33d4bb7e34206a86eb1263c38fb8d"}]}
```

All of these operations have the expected result.

---

The issue template asks you to _"Please attach (or link to) screenshots and logs if applicable"_, but I couldn't find any in the reported issue. Please comment with these so that we can investigate further.

The issue template asks you about _"Your environment: OS/Browser/Gateway/Device/...? Versions?"_, but I couldn't find any information in the reported issue. Please comment with this so that we can investigate further.

pdubois22, 2019-02-27:
It is common practice to return an HttpStatus code of 401 when credentials are wrong:

Correct key, correct device: (note the -i in curl params)

```
curl -i -H "Authorization: Bearer NNSXS.SH6GDJWLELTTU725WJH5UABYJQ4AKGV7I4W6LLI.ZEUKPV3D2KHTVQRLQPNGQWXQ6PO4QFF6ZU4BTM5INXXEFZENWHNA" "localhost:1885/api/v3/applications/caf/devices/850706c76703591801670cef9303200b"
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Content-Type: application/json
Vary: Origin
X-Content-Type-Options: nosniff
X-Frame-Options: SAMEORIGIN
X-Request-Id: 01D4QHG6QZZ79PK26KSYSJMMEM
X-Xss-Protection: 1; mode=block
Date: Wed, 27 Feb 2019 13:15:02 GMT
Content-Length: 255

{"ids":{"device_id":"850706c76703591801670cef9303200b","application_ids":{"application_id":"caf"},"dev_eui":"0018B20000002642","join_eui":"70B3D59520000000"},"created_at":"2019-02-27T09:22:28.397Z","updated_at":"2019-02-27T09:22:28.428Z","version_ids":{}}

```

Incorrect Key in auth / Bearer:
```
curl -i -H "Authorization: Bearer MFRWG.FKSG4PLAMFSTAUF7UWG45MGT4PRG3IBDV2NUVCA.OR6AK5JGXQLOMY7L6ASV3KINDMBUB4RW32OJUO2KUE6E6SL" "localhost:1885/api/v3/applications/caf/devices/850706c76703591801670cef9303200b"
HTTP/1.1 404 Not Found
Access-Control-Allow-Origin: *
Content-Type: application/json
Vary: Origin
X-Content-Type-Options: nosniff
X-Frame-Options: SAMEORIGIN
X-Request-Id: 01D4QHK857Q2GWSM5JP3DS981G
X-Xss-Protection: 1; mode=block
Date: Wed, 27 Feb 2019 13:16:42 GMT
Content-Length: 354

{"code":5,"message":"error:pkg/identityserver/store:access_token_not_found (access token not found)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/identityserver/store","name":"access_token_not_found","message_format":"access token not found","attributes":{},"correlation_id":"a86b9de7cfc14ed99d024cba42dd476d"}]}
```

**Could (should?) be:** 

```
curl -i -H "Authorization: Bearer MFRWG.FKSG4PLAMFSTAUF7UWG45MGT4PRG3IBDV2NUVCA.OR6AK5JGXQLOMY7L6ASV3KINDMBUB4RW32OJUO2KUE6E6SL" "localhost:1885/api/v3/applications/caf/devices/850706c76703591801670cef9303200b"
HTTP/1.1 401 Unauthorized
Access-Control-Allow-Origin: *
Content-Type: application/json
Vary: Origin
X-Content-Type-Options: nosniff
X-Frame-Options: SAMEORIGIN
X-Request-Id: 01D4QHK857Q2GWSM5JP3DS981G
X-Xss-Protection: 1; mode=block
Date: Wed, 27 Feb 2019 13:16:42 GMT
Content-Length: 354

{"code":5,"message":"error:pkg/identityserver/store:access_token_not_found (access token not found)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/identityserver/store","name":"access_token_not_found","message_format":"access token not found","attributes":{},"correlation_id":"a86b9de7cfc14ed99d024cba42dd476d"}]}
```

PR #211 (2019-02-28): Fix error codes for failed auth
**Summary:**

This PR closes #197 by returning `Unauthenticated` error types if API keys, OAuth tokens or their corresponding user/client is not found.

**Changes:**

- Return `Unauthenticated` errors if entities were not found during Auth procedure
- Return `Unauthenticated` errors if hash validation fails (instead of `PermissionDenied`)
closed 2019-02-28 by htdvisser