TheThingsNetwork/lorawan-stack #202: AWS 'Appliance' for Stack V3

mrpackethead, 2019-02-27:
**Summary:** 

Creation of a community AMI image for deployment in Amazon EC2.  ( once its stable )

...

**Why do we need this?**
Very quick and error free deployment 

...

**What is already there? What do you see now?**

...

**What is missing? What do you want to see?**

...

**How do you propose to implement this?**

...

**Environment:**

...

**What can you do yourself and what do you need help with?**

...

johanstokking, 2019-02-27:
Thanks for raising this. Even though we do understand that individual machine images for commercial cloud platforms come in handy, the variety of distributions of this software stack is endless.

We have decided to distribute the sources, binaries for various operating systems and architectures, Docker images as well as packages, supporting a wide variety open source targets, namely:
- Brew - #140
- APT - #140
- RPM - #140
- Snap - #140 (publish to Snapcraft #114)
- AUR - #110 
- Nixpkgs - #111 

Adding machine images to this list involves a lot of work, testing, maintenance and community support. Also, it's hard to find a one-size-fits-all solution considering the number of parameters (underlying OS, networking, storage, etc). Therefore, we decided against these distributions, and we will not support the community in doing so.

Finally, when deploying a private network on a commercial cloud platform, you probably want to use their managed databases (instead of a local Postgres/Cockroach and Redis), their IoT platform as device registry and pub/sub, as well as be ready to scale (i.e. not a single vm, but ideally a managed Kubernetes cluster with their load balancer with TLS termination, etc). This is exactly what The Things Industries will be offering soon, for the major commercial cloud platforms.

closed 2019-02-27 by johanstokking

mrpackethead, 2019-02-27:
Yes, fair enough,