TheThingsNetwork/lorawan-stack #206: API call not simple to use

pdubois22, 2019-02-27:
**Summary:** 
I am trying to implement the following CLI call:

dist/linux_amd64/ttn-lw-cli devices create caf dev3 --defaults=true \
--supports-join=true \
--join-eui=70B3D59520000000 --dev-eui=A81758FFFE031BF9 \
--root-keys.app-key.key=6E2DD93BA3DF1BE62D57C33D756594E4 \
--session.dev-addr=FC006A0C \
--session.keys.app-s-key.key=2412B7B219FCEE78BDFACB5944764116 \
--session.keys.f-nwk-s-int-key.key=57A976DF984AF3EC514B21E56C081510 \
--lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
--frequency-plan-id EU_863_870 \
--session.last-f-cnt-up=25860 \
--session.last-n-f-cnt-down=901

From what I understand, I need at least 3 http post:
one to the is:
/api/v3/applications/xxx/devices

one to the ns:
/api/v3/ns/applications/xxx/devices

one to the js:
/api/v3/js/applications/xxx/devices

and maybe one to the as ?
one to the ns:
/api/v3/as/applications/xxx/devices

Each call requires a subset of the info above and appropriate field_mask.paths. Correct ?

**Why do we need this?**
Simplify the API call or at least a clear documentation.

**What is already there? What do you see now?**
haven't seen much.

**What is missing? What do you want to see?**
a simple / single API to create / update devices.
If not possible because of the separation of components, at least some documentation about it.

**How do you propose to implement this?**
Ideally a single / simple call : http post or put.

**Environment:**
API call

**What can you do yourself and what do you need help with?**
I am not confortable enough with the components to write a go api, but I can try to write the doc when I sorted this out / or somebody has clarified the proper way to use the API.

johanstokking, 2019-02-27:
>If not possible because of the separation of components, at least some documentation about it.

We should document this better; a documentation system on the way.

Indeed, this stack is designed for distribution, with potentially different domains for JS, NS and AS. Only the client (API, CLI, Console) and ER (entity registry, exposed by IS) know what the JS, NS and AS endpoints are, but only the client has access in terms of security context as well as network access. For example, we can't call AS device registry from the community ER if that AS is hosted behind a firewall.

We are making this much easier for developers in the SDKs.

johanstokking, 2019-02-27:
See https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/ttn-lw-cli/commands/end_devices_split.go for how the CLI does it currently.

#167 is going to enforce this in the APIs.

johanstokking, 2019-09-11:
@Sypheos please make this part of #1118

PR #1741 (2019-12-20): Describe EndDevice fields stored by IS/NS/AS/JS

#### Summary

This PR extends the EndDevice API reference with an explanation about how end devices are stored in multiple registries and adds details on (some of) the fields stored by each registry.

Closes #206

#### Notes for Reviewers

This is not intended to be complete, just enough to get started with.

Feel free to add commits to this PR for anything essential that is still missing.
closed 2019-12-20 by htdvisser