TheThingsNetwork/lorawan-stack #214: Application Link management

bafonins, 2019-03-01:
**Summary:** 

Original umbrella issue: https://github.com/TheThingsNetwork/lorawan-stack/issues/24
Add support for linking applications to the NS.

**Why do we need this?**

This is needed for managing the way the AS links to the NS for traffic exchange. 

**What is already there? What do you see now?**

- The applications section in the console
- API for linking

**What is missing? What do you want to see?**

The application link subsection in the console, linking methods in the sdk, respective functionality on the frontend:
1. Create link
2. Edit link
3. Remove link

**How do you propose to implement this?**

A form is required with the following fields (bare minimum):
- the network server address
- the api key, requires https://github.com/TheThingsNetwork/lorawan-stack/issues/142 to be merged

And the `unlink` button if the link is present.

**Environment:**

Browser

**What can you do yourself and what do you need help with?**

Implement it

cc @kschiffer

kschiffer, 2019-03-06:
I understand there might be some expectation from the user that this linking process will be done automatically upon creating the application (as it is now in v2).
I know we talked about this process regarding avoiding to carelessly do things automatically, but wouldn't it make sense to add a checkbox in the create new application form ("Also link application to network server") which would be checked by default? This is likely what a vast majority of users would want to do right?
What do we gain from making this step explicit only, @htdvisser?

htdvisser, 2019-03-06:
We have to implement the explicit management anyway. The implicit flow would only be a convenience so that users don't have to create an API key and copy-paste that into the link configuration.

The reason I think we should only implement the explicit steps now, and implement the implicit flow later is simple: time.

johanstokking, 2019-03-06:
Let's consistently keep calling this `application upstream messages` (i.e. join-accept and uplink messages with FPort > 0) and `application downlink messages`. I'm fine omitting `application` if that is clear in this context.

closed 2019-03-12 by bafonins