TheThingsNetwork/lorawan-stack #217: Fix API definition mismatch

bafonins, 2019-03-04:
**Summary:** 

The `api-definition.json` file is not synced with the current api definitions of the stack.
See https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/applicationserver.proto#L71 and https://github.com/TheThingsNetwork/lorawan-stack/blob/master/sdk/js/generated/api-definition.json#L188 as an example.

See tech dept issue: https://github.com/TheThingsNetwork/lorawan-stack/issues/32

**What do you see now?**

Deviation in the generated definitions from the actual api.

**What do you want to see instead?**

No deviation.

**How do you propose to implement this?**

Use `sdk/js/src/util/http-mapper.js` to update the api definition.

**Environment:**

Javascript SDK

**What can you do yourself and what do you need help with?**

Fix it

htdvisser, 2019-03-04:
Generating the these files needs to become part of our build tooling. Let's discuss what exactly you need, because it doesn't look like you're using both `api.json` and the derived `api-definition.json`.

bafonins, 2019-03-04:
> Generating these files needs to become part of our build tooling.

This is already specified here https://github.com/TheThingsNetwork/lorawan-stack/issues/30 in the `CI` section.

To clarify, currently, we parse the `generated/api.json` file to extract the data (http method, url, parameters, service name e.g. `ApplicationAccess`, service method names e.g.`CreateAPIKey`) that is needed to generate all stack api methods.

I would say that any altering of protos should also update the definitions that we have at the moment together with the markdown/swagger docs. Ff we decide to go with https://github.com/TheThingsNetwork/lorawan-stack/issues/32, then it should update the `es6` classes as well.

@kschiffer could you clarify how you updated the `api-definition.json` when renaming devices?

kschiffer, 2019-03-04:
To update `api-definition.json`, run `sdk/js/dist/util/http-mapper.js`. This will ingest `api.json` and generate `api-definition.json` from it.

bafonins, 2019-03-04:
@kschiffer what about the `generated/api.json` file?

kschiffer, 2019-03-04:
This is what I thought gets generated from `make protos`, but apparently not. @htdvisser, can you clarify? I did not work on the generator for this.

htdvisser, 2019-03-04:
As I wrote in https://github.com/TheThingsNetwork/lorawan-stack/issues/217#issuecomment-469205560, this still needs to become part of our build tooling. In order to implement that, I'd need to know what exactly you need and what not.

bafonins, 2019-03-04:
@kschiffer how did you generate the file for initial commit for the js sdk?

kschiffer, 2019-03-04:
It was provided to me by @htdvisser. It used to be generated with this command apparently
```
rm -rf tmp
mkdir tmp
docker run --rm -v $PWD:$PWD -w $PWD htdvisser/protocontainer -I $(dirname $PWD) -I $GOPATH/src --doc_out=$PWD/tmp --doc_opt=json,api.json --swagger_out=$PWD/tmp $PWD/api/*.proto
```
which does not seem to work anymore.

johanstokking, 2019-03-26:
Closed by #231

closed 2019-03-26 by johanstokking