TheThingsNetwork/lorawan-stack #230: Implement LoRaWAN roaming

johanstokking, 2019-03-05:
**Summary:** 

Implement LoRaWAN hand-over and passive roaming.

**Why do we need this?**

That is a very good question. One reason would be LoRaWAN Backend Interfaces compliancy, but;

- Hand-over roaming: I don't see a market for this
- Passive roaming
   - Stateless (simple): this may be useful at some point
   - Stateful (device session): I don't see a market for this (at this point)

**What is already there? What do you see now?**

Preparation for implementing LoRaWAN Backend Interfaces; separation of component roles, API forward compatibility, etc.

**What is missing? What do you want to see?**

Hand-over and passive roaming support.

**How do you propose to implement this?**

Implement LoRaWAN Backend Interfaces 1.1 draft 3 or higher.

**What can you do yourself and what do you need help with?**

Design, implement or review.

**Discussion**

First of all, we should consider whether we should do this in the first place.

My proposal would be to;

1. Not implement the Backend Interfaces in the core components for the time being. If we do implement it, then first stateless passive roaming, then stateful, then hand-over roaming;
2. Keep potential future support for it in mind, as we did so far with V3;
3. Implement stateless passive roaming support around the Packet Broker, i.e. not in the Network Server

nejraselimovic, 2021-05-12:
@johanstokking I remember you were mentioning something about LoRaWAN roaming recently, so what is the status with that? Is that something we are aiming for in the following months?

johanstokking, 2021-05-12:
Thanks for raising this. We went for option 3:

> 3\. Implement stateless passive roaming support around the Packet Broker, i.e. not in the Network Server

Packet Broker now support LoRaWAN stateless passive roaming. We're doing some operations and interoperability testing work still.

References https://github.com/packetbroker/router/pull/62
References https://github.com/packetbroker/infrastructure/issues/32
References https://github.com/TheThingsNetwork/lorawan-stack/pull/4151

So we can close this.

closed 2021-05-12 by johanstokking