TheThingsNetwork/lorawan-stack #233: Full page errors

kschiffer, 2019-03-05:
**Summary:** 
As a result of #41, we need a proper error page that will be shown when fatal errors occur that prevent the app from loading. This error page should match the static ones that are rendered by the backend (https://github.com/TheThingsIndustries/lorawan-stack/issues/1315#issuecomment-440804413).

A request `curl -H "Accept: text/html" -XPOST http://localhost:1885/oauth/api/auth/login` results in the following (formatted for readability):

```html
<!doctype html>
<html lang="en">

<head>
    <!-- ... -->
</head>

<body>
    <div id="app"></div>
    <script>
        window.APP_ROOT = "/oauth";
        window.ASSETS_ROOT = "http://localhost:1885/assets";
        window.APP_CONFIG = { "language": "en" };
        window.PAGE_DATA = {
            "error": {
                "code": 2,
                "message": "error:pkg/errors/web:http (HTTP error: code=403, message=invalid csrf token)",
                "details": [
                    {
                        "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
                        "namespace": "pkg/errors/web",
                        "name": "http",
                        "message_format": "HTTP error: {message}",
                        "attributes": {
                            "message": "code=403, message=invalid csrf token"
                        }
                    }
                ]
            }
        };
    </script>
    <script type="text/javascript" src="http://localhost:1885/assets/oauth.b4ebca19adbb1b50d911.js"></script>
</body>

</html>
```

Errors are structured in the same way as the errors you receive from the API, so with the `details` array, where you need to find the object with `"@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails"` to get TTN error details.

**Why do we need this?**
Because the errors returned from the backend need to be properly displayed to the user.

**What is already there? What do you see now?**
We have a screen design ready:
[image]
This should cover most cases as it can display: Error Code, Error Name, Error Message.
My only concern would be the state of the header/nav-bar. Depending on the situation it could make sense to have it rendered fully (with links and user info) and in others it might not be possible. I will come up with something here.

**What is missing? What do you want to see?**
A logic in the console/oauth app that checks for the existence of `PAGE_DATA.error` and displays the error respectively.

**How do you propose to implement this?**
We could use the common init component (`lib/components/init.js`) to take care of this. The component could make the check and pass possible errors to an error view component that would need to be built with styling according to the screen design above.

**What can you do yourself and what do you need help with?**
This should be pretty straight-forward.

pierrephz, 2020-03-04:
Hey, here is a 404 error page that can be used as a template for other error messages in the console

[image]

here is the code, its already responsive and ready to use (just need to be edited with values we want for each error we have)

[error_page.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4287640/error_page.zip)

Open for feedbacks and modifications

(bonus: dark mode, if we decide to go this way one day, this is never waste ⚫️[error_page_dark.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4287651/error_page_dark.zip))

kschiffer, 2020-09-21:
So from a technical perspective, this is already implemented as defined in the OP. Our outermost error boundary will catch errors and render the error view. I will close this issue and open a new one about static HTML/CSS templates for backend error responses on routes that are not handled by our frontend react apps.

https://github.com/TheThingsNetwork/lorawan-stack/issues/3261

closed 2020-09-21 by kschiffer