TheThingsNetwork/lorawan-stack #235: Support for maps

kschiffer, 2019-03-05:
**Summary:** 
The console needs to be able to show the location of [gateway](#26) (antennas) and [devices](#25) on a map. We need hence to decide which technology to use for this.

**Why do we need this?**
To display locations properly and conveniently.

**What is already there? What do you see now?**
Nothing of that sort.

**What is missing? What do you want to see?**
- A react component that we can feed with location(s) and which will display them in a map
- A way to also set and modify locations through this component

**How do you propose to implement this?**
My initial impulse would be building this on top of [leaflet.js](https://leafletjs.com/) but it sure makes sense to discuss the options here briefly. I personally dig [mapbox](https://www.mapbox.com/) a lot, which has extensive features, but it is [not free](https://www.mapbox.com/pricing/).

**What can you do yourself and what do you need help with?**
Input welcome, we can then follow up more precisely.

bafonins, 2019-03-05:
Isnt mapbox just the tile provider used by leaflet? Correct if im wrong.
Im with using leaflet. Had some experience with it before and it seems to cover all our needs at the moment.

kschiffer, 2019-03-05:
Yes correct. It would then be just an API layer on top of raster tiles provided by mapbox, giving us the flexibility to switch the tile provider later on easily. From what I see, the difference technically (apart from a different API) would be that leaflet uses raster tiles, while native mapbox is capable of vector tiles. Vector tiles seem to be nice for advanced use cases which we will likely not use anytime soon.
So if you have some experience with leaflet already then I think we should use that indeed.

closed 2019-09-18 by mjamescompton