TheThingsNetwork/lorawan-stack #245: Add missing field validations

rvolosatovs, 2019-03-06:
**Summary:** 

Currently there is a bunch of fields in our API, which lack validation options.
E.g. https://github.com/TheThingsNetwork/lorawan-stack/blob/4f5dc70c4a3068aa2169d8a8e69d7c99507294db/api/end_device.proto#L144-L147
Refs #51 

**Why do we need this?**

To make sure we don't store garbage

**What is already there? What do you see now?**

Possible garbage in fields

**What is missing? What do you want to see?**

Validation

**How do you propose to implement this?**

Add the validators.
For the exact case linked I propose `(validate.rules).enum.defined_only = true`
and `(validate.rules).enum.not_in = [0]`
See https://github.com/lyft/protoc-gen-validate#enums

rvolosatovs, 2019-03-06:
Blocked by #194

rvolosatovs, 2019-03-15:
I added a bunch of missing validations in #194, however those were mostly NS or JS-related and obvious ones, like email.
Let's go component-by-component and ensure that all the fields that need validation are validated.
NS and JS should be fine, however we should still look through GS, AS and IS. I'll assign package maintainers.
@johanstokking 	please check AS and GS-related messages
@htdvisser 	please check IS-related messages

htdvisser, 2019-03-18:
Unblocked since #194 was merged

johanstokking, 2019-03-18:
@rvolosatovs checked both, only addition necessary seems to be in #297

PR #341 (2019-03-25): Identity Server (and some other) field validations
**Summary:**

This PR addresses the Identity Server part of the missing field validations, and thereby closes #245.

**Changes:**

- Validation rules for entity names (max len 50), descriptions (max len 2000) and attribute map keys (TTN ID pattern, this affects the implementation for #75 @adriansmares)
- Validation of repeated enum fields for defined values
- Validation of URL fields
- Validation of IDs
closed 2019-03-25 by htdvisser