TheThingsNetwork/lorawan-stack #263: Set up LoRaWAN Device Repository

htdvisser, 2019-03-08:
**Summary:** 

We should create a LoRaWAN device repository in TheThingsNetwork's Github organization.

**Why do we need this?**

The device respository will contain descriptions and definitions of well-known LoRaWAN devices, including the default MAC parameters the devices reset to, the payload formats, and possibly other information used by the NS, AS (and JS).

**What is already there? What do you see now?**

We already have `pkg/devicerepository` implemented using `pkg/fetch`. The repository can be configured with the `--device-repository.url` or `--device-repository.directory` flags. The AS already uses it (if configured) for payload formatters.

**What is missing? What do you want to see?**

There is no default repository on our Github that we can configure.

**How do you propose to implement this?**

Create repository `TheThingsNetwork/lorawan-devices` and use that as default in the Stack.

**What can you do yourself and what do you need help with?**

I think it would be best if @johanstokking would set up the repository, add the README and maybe some examples.

johanstokking, 2019-06-21:
Note to self; https://github.com/Azure/IoTPlugandPlay/tree/master/DTDL

johanstokking, 2020-11-03:
Device repository is now setup; https://github.com/TheThingsNetwork/lorawan-devices.

Will file follow-up issues on implementing this in the Console for onboarding devices.

johanstokking, 2020-11-03:
Closed by https://github.com/TheThingsNetwork/lorawan-devices/pull/11

closed 2020-11-03 by johanstokking