TheThingsNetwork/lorawan-stack #265: User Session Management

htdvisser, 2019-03-08:
**Summary:** 

The Identity Server should support management of user sessions.

**Why do we need this?**

Although most of our stack uses OAuth access tokens, the OAuth server itself uses user sessions. Users may want to list and possibly revoke these sessions.

**What is already there? What do you see now?**

The [User Session Store](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/identityserver/store/store_interfaces.go#L97-L106) supports listing and deleting sessions. 

**What is missing? What do you want to see?**

- An API for listing/deleting user sessions
- This API to be exposed by the Identity Server

**How do you propose to implement this?**

Pretty similar to #264

**What can you do yourself and what do you need help with?**

I can do this.

htdvisser, 2019-07-09:
Updated title and description. @johanstokking 	@kschiffer 	comments please.

johanstokking, 2019-07-10:
I think it's good to work towards a screen like that, but I see a complicated solution for a simple problem.

That is, if you logout in an OAuth client (i.e. CLI or Console), and you log back in, the OAuth server logs you in automatically, so you don't have a chance to switch accounts.

Right?

Isn't that fixed by the switcher in https://github.com/TheThingsNetwork/lorawan-stack/issues/488 ?

kschiffer, 2019-07-11:
Note that google apps essentially solve this by integrating the account switcher into their apps:
[image]
It also means that clicking logout there will actually log you out from your google account altogether, which is in line with what users likely expect to happen. I think a solution like this would be very nice to have in the console because it is very clear to the user and would solve the login/logout confusion, because people will have a hard time understanding the separation of oauth server and console app. So from a user perspective, this will tie the two closer together.
However, I can't assess whether this solution would be possible with our current setup.

So this is a solution for "first-class" applications like the console. Third party apps would still need to rely on a dedicated account switcher.

johanstokking, 2019-07-11:
I agree that having the OAuth client logout actually go to the OAuth server is a shortterm and effective solution.

johanstokking, 2019-07-23:
So what do we decide here?

htdvisser, 2019-07-24:
I think we should do (in order)

- 1. After console logout, redirect the user to a logout page on the OAuth/Account UI, something like `/oauth/logout`, but we may want to look at how other systems such as OIDC do this (this issue)
- 2. Before the OAuth Authorize, we should add a screen that asks the user if they want to continue as <account> (#488)
- 3. On the OAuth/account UI, show where else a user has valid OAuth access tokens (other CLI/Console/... logins) and allow the user to revoke those too (this issue).

johanstokking, 2019-07-24:
OK sounds good.

kschiffer, 2019-07-24:
> After console logout, redirect the user to a logout page on the OAuth/Account UI, something like /oauth/logout

So this would be a simple page with a logout button?

htdvisser, 2019-07-24:
1. For the MVP yes, it would show who is logged in, and a logout button. Like what we have now, but then fancier.

2. Yes, for every client that is pre-authorized or previously authorized by the user, the screen is skipped. This is per spec. There would indeed be an extra screen "continue as <user>" before the authorize screen (which is still skipped for already authorized clients)

htdvisser, 2019-10-08:
Replaced by #1422.

closed 2019-10-08 by htdvisser