TheThingsNetwork/lorawan-stack #267: Revoke all User Sessions and OAuth Access Tokens on password change

htdvisser, 2019-03-08:
**Summary:** 

Users should get the possibility to revoke all User Sessions and OAuth Access Tokens on a password change.

**Why do we need this?**

In case a user login gets compromised we need to be able to revoke access.

**What is already there? What do you see now?**

We have the password reset flow, we'll have #264 and #265.

**What is missing? What do you want to see?**

- A `revoke_access` boolean in `UpdateUserPasswordRequest` to indicate that existing access should be revoked.
- Code in the IS's UpdatePassword handler to revoke sessions and access tokens if the user wants that.

**How do you propose to implement this?**

First we need to be able to list user sessions (#265) and access tokens (#264) in the stores. Then we'll update the `UpdateUserPasswordRequest`, and add the delete logic to the UpdatePassword handler.

**What can you do yourself and what do you need help with?**

I can do this a soon as #264 and #265 are done

furtiman, 2019-07-23:
@htdvisser I recall you wanted to format this issue in some way?

htdvisser, 2019-07-24:
Updated

furtiman, 2019-07-24:
@htdvisser 

> Add a --revoke_all_access flag to ttn-lw-cli logout

You meant `ttn-lw-cli user update-password`, right?

htdvisser, 2019-07-24:
Yes, that's what I meant. Updated.

PR #1058 (2019-07-26): Revoke sessions and access tokens on password change

#### Summary

Closes #267 

#### Changes

- Added a boolean field `RevokeAllAccess` to `UpdateUserPasswordRequest`
- Added is logic to revoke tokens and sessions if the field is true
- Added boolean cli flag to `user` `update-password`

#### Notes for Reviewers

I have tested the new flag, and if it is set upon password change, the console session logs out automatically, requesting a new login.

Changing the password without the flag lets you stay in the console after changing the password.
closed 2019-07-26 by furtiman