TheThingsNetwork/lorawan-stack #268: Application Collaborators management

bafonins, 2019-03-08:
**Summary:** 

The console should provide functionality for managing collaborators of an application.
Umbrella issue: https://github.com/TheThingsNetwork/lorawan-stack/issues/27

**What is already there? What do you see now?**

- Screen design 
[image]
- The Applications section in the console
- The API for Application Collaborators
  - `ListCollaborators` (`GET /applications/<application-id>/collaborators`) - enumerate collaborators in the **Current Collaborators** section
  - `SetCollaborator` (`PUT /applications/<application-id>/collaborators`) - submitting the form in the **Add Collaborator** section and delete a collaborator (user/organization) in in the list. 
- The API for Application Rights
  - `ListRights` (`GET /applications/<application-id>/rights`) - application rights to be displayed in the **Define rights** subsection.

**What is missing? What do you want to see?**

- Application Collaborators methods in the SDK
- The "Collaborators" subsection of Application in the console.
- Search API for Application Collaborators
- Search API for Users/Organizations (to allow the users to choose the exact entity to be added as a collaborator)
- 

**How do you propose to implement this?**

...

**Environment:**

Browser

**What can you do yourself and what do you need help with?**

- Discuss the missing functionality
- Extend the Applications service in the SDK
- Implement UI in the console

cc @kschiffer @johanstokking @htdvisser  for comments

htdvisser, 2019-03-08:
Regarding the screendesign:

- I don't know how you would realistically determine the emphasis in "Manage the application **settings**", especially not with translations of the console.
- The "Rights" column in the collaborators list contains some short version of right descriptions. Those short versions aren't defined anywhere.
- The "Rights" column in the collaborators list only shows a couple of rights. I don't know how realistic that is. Collaborators will either have all rights, or have a selection of rights (on organizations this could be many).

Why does it look completely different from the (almost identical) API key feature https://github.com/TheThingsNetwork/lorawan-stack/pull/171 ?

Regarding the desired functionality:

- Searching for existing collaborators within the entity is definitely not MVP
- I think MVP should also work without the search/autocomplete of accounts (users/organizations) in the network

bafonins, 2019-03-08:
> I don't know how you would realistically determine the emphasis in "Manage the application settings", especially not with translations of the console.

This is relative old screen design. This wont the case.

> The "Rights" column in the collaborators list contains some short version of right descriptions. Those short versions aren't defined anywhere.

Again. Old screen design.

kschiffer, 2019-03-12:
Indeed, looking at the screen design again, it was based on a couple of assumptions. Let's stick with an implementation complementary to the API Keys feature for now. I, however, like that how the screen design enables adding and viewing collaborators on one screen.

bafonins, 2019-03-26:
1. Do we want to allow the users to edit their application rights as collaborators?
2. Should the users be able to remove themselves from the list of collaborators?
  2.1 If yes, how this should look ui wise?

Sypheos, 2019-03-26:
Those feature comes in from time to time on support.
1. If they have the right to manage collaborators' rights.
2. Yes, as a hop out button with some double checking. If they are the last person having rights on the application, send them to the delete screen.

Some use cases, involve the a third party registering the application and devices for a client. Once the client is onboard, The third party then need to be remove himself from the list of collaborators. This could be done by the client but the experience would be different.

htdvisser, 2019-03-26:
For (1) indeed what @Sypheos says. If the user has the right to manage collaborators (`RIGHT_<ENTITY>_SETTINGS_COLLABORATORS`) they can edit the rights of all collaborators, including themselves. One thing to keep in mind when **editing** rights of existing collaborators is that the backend will refuse to set rights that the caller does not have. Not sure if/how we should handle that UI-wise.

Regarding (2) I think the simplest solution is to add a "revoke all rights" button (this is the same as "delete collaborator") to the "edit collaborator" view. 

The console is **not** responsible for making sure that entities have at least one admin/collaborator, so disregard that part of @Sypheos's message.

The use case of a collaborator *without* the right to manage collaborators that wants to remove their own rights is for later concern, don't worry about that. The user (support agent) in the use case that @Sypheos mentions should just be given the rights to manage collaborators until that customer is fully onboarded.

Sypheos, 2019-03-26:
> The console is not responsible for making sure that entities have at least one admin/collaborator

Then what happens ? The network send an error or is the application orphaned and the admin get access ?

htdvisser, 2019-03-26:
I think right now it's the latter

bafonins, 2019-03-27:
blocked by https://github.com/TheThingsNetwork/lorawan-stack/issues/353

bafonins, 2019-03-27:
>> One thing to keep in mind when editing rights of existing collaborators is that the backend will refuse to set rights that the caller does not have. Not sure if/how we should handle that UI-wise. 

@kschiffer

bafonins, 2019-03-27:
nvm, it is already handled by the backend
[image]

bafonins, 2019-03-27:
@htdvisser what about when the users organization is a collaborator for an app?

consider the following scenario:
1. `test-user` has an application (`test-user-app`)
2. `test-user2` has an organization (`test-user2-org`)
3. `test-user` adds `test-user2-org` as a collaborator for `test-user-app`
4. `test-user2` has no access to the list of collaborators of the `test-user-app` application (same for editing/linking/api-keys management)

```
// as test-user:
bafonins > curl -H "Authorization: Bearer $access_token" -XPOST --data '{"application": {"ids": {"application_id": "test-user-app"}}}' http://localhost:1885/api/v3/users/test-user/applications | jq  
{
  "ids": {    "application_id": "test-user-app"  },  "created_at": "2019-03-27T13:11:21.693Z",
  "updated_at": "2019-03-27T13:11:21.693Z"
}

// as test-user2:
bafonins > curl -v -H "Authorization: Bearer $access_token_2" -XPOST --data '{"organization": {"ids": {"organization_id": "test-user2-org"}}}' http://localhost:1885/api/v3/users/test-user2/organizations | jq
{
  "ids": {
    "organization_id": "test-user2-org"
  },
  "created_at": "2019-03-27T13:15:43.771Z",
  "updated_at": "2019-03-27T13:15:43.771Z"
}

// as test-user:
bafonins > curl -v -H "Authorization: Bearer $access_token" -XPUT --data '{"collaborator": {"ids": {"organization_ids": {"organization_id": "test-user2-org" }}, "rights": ["RIGHT_APPLICATION_ALL"]}}}' http://localhost:1885/api/v3/applications/test-user-app/collaborators | jq
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying ::1...
* upload completely sent off: 125 out of 125 bytes
< HTTP/1.1 200 OK
< Access-Control-Allow-Origin: *
< Access-Control-Expose-Headers: Link, Date, Content-Length, X-Total-Count
< Content-Type: application/json
< Vary: Origin
< X-Content-Type-Options: nosniff
< X-Frame-Options: SAMEORIGIN
< X-Request-Id: 01D6ZMY12EN1HPS4BE5M3V487G
< X-Xss-Protection: 1; mode=block
< Date: Wed, 27 Mar 2019 13:20:20 GMT
< Content-Length: 2
{}

// as test-user:
bafonins > curl -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/applications/test-user-app/collaborators | jq
{
  "collaborators": [
    {
      "ids": {
        "user_ids": {
          "user_id": "test-user"
        }
      },
      "rights": [
        "RIGHT_APPLICATION_ALL"
      ]
    },
    {
      "ids": {
        "organization_ids": {
          "organization_id": "test-user2-org"
        }
      },
      "rights": [
        "RIGHT_APPLICATION_ALL"
      ]
    }
  ]
}

// as test-user2:
bafonins > curl -H "Authorization: Bearer $access_token_2" http://localhost:1885/api/v3/applications/test-user-app/collaborators | jq  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current                                 Dload  Upload   Total   Spent    Left  Speed
100   385  100   385    0     0   5661      0 --:--:-- --:--:-- --:--:--  5746
{
  "code": 7,
  "message": "error:pkg/auth/rights:no_application_rights (no rights for application `test-user-app`)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/auth/rights",
      "name": "no_application_rights",
      "message_format": "no rights for application `{uid}`",
      "attributes": {
        "uid": "test-user-app"
      },
      "correlation_id": "63e4eb046ea84d84959e7a770176214d"
    }
  ]
}
```

bafonins, 2019-04-01:
blocked by https://github.com/TheThingsNetwork/lorawan-stack/issues/422

closed 2019-04-17 by bafonins