TheThingsNetwork/lorawan-stack #271: Downlink fails because too many MAC requests pending

pdubois22, 2019-03-08:
**Summary:** 
When too many requests are pending, the buf may exceed the capability of the payload for the SF.

**Steps to Reproduce:** 
Not so easy, but the problem is pkg/networkserver/downlink.go:139 and after:
the cmdBuf grows with all dev.MACState.PendingRequests without a consideration for size.

**What do you see now?**
Unable to schedule downlink in this case.

```
DEBUG Performing MIC checks on devices with matching frame counters... dev_addr=FC006A95 device_count=1 device_uid=caf.850706c76703591801670ce770f41899 full_f_cnt_up=3 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs last_f_cnt_up=2 namespace=grpc payload_length=15 request_id=01D5EDPXJX0EB9SG9V74VQ7NBY uplink_f_cnt=3 uplink_f_cnt_up=3
 DEBUG Matched device                           dev_addr=FC006A95 device_uid=caf.850706c76703591801670ce770f41899 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D5EDPXJX0EB9SG9V74VQ7NBY
  WARN Application Server not found, not forwarding uplink dev_addr=FC006A95 device_uid=caf.850706c76703591801670ce770f41899 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D5EDPXJX0EB9SG9V74VQ7NBY
 DEBUG Waiting for collection window to be closed... grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D5EDPXJX0EB9SG9V74VQ7NBY
 DEBUG Processing downlink task...              delay=1.077ms device_uid=caf.850706c76703591801670ce770f41899 start_at=2019-03-08 10:31:20.616873 +0000 UTC
 DEBUG Generated downlink                       ack=true delay=1.077ms device_uid=caf.850706c76703591801670ce770f41899 f_cnt=1 f_pending=false f_port=0 m_type=UNCONFIRMED_DOWN mac_count=10 payload_length=67 start_at=2019-03-08 10:31:20.616873 +0000 UTC
 DEBUG Scheduling downlink...                   attempt_rx1=true attempt_rx2=true delay=1.077ms device_class=CLASS_A device_uid=caf.850706c76703591801670ce770f41899 downlink_class=CLASS_A downlink_type=data path_count=1 rx1_delay=5 rx1_frequency=867100000 rx2_data_rate=0 rx2_frequency=869525000 start_at=2019-03-08 10:31:20.616873 +0000 UTC
 DEBUG Scheduling downlink                      class=CLASS_A gateway_eui=00000008004A3372 gateway_uid=eui-00000008004a3372 namespace=gatewayserver/io/udp
 DEBUG Failed to schedule on path               error=error:pkg/gatewayserver/io:too_long (the payload length `67` exceeds maximum `` at data rate index `0`) gateway_uid=eui-00000008004a3372 grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc request_id=01D5EDPXSBYXBKKJDZCT6MF1QM

```
**What do you want to see instead?**
I am not so sure. limit the requests included to avoid the error for sure. oldest first?

**How do you propose to implement this?**
fix downlink.go

**Environment:**
master

**What can you do yourself and what do you need help with?**

...

htdvisser, 2019-03-08:
Thanks for reporting this, @pdubois22. If I remember correctly I've heard @rvolosatovs about this before, and there was some idea to prioritize the MAC commands and split them over multiple messages.

@johanstokking it also looks like the `too_long` error misses the attribute for maximum:
```
error:pkg/gatewayserver/io:too_long (the payload length `67` exceeds maximum `` at data rate index `0`)
```

johanstokking, 2019-03-08:
Indeed we should prioritize and clip the MAC commands.

Error attributes is fixed in #275.

rvolosatovs, 2019-03-13:
NS already only schedules as many MAC commands as can be sent and answered in one frame, in compliance with the LoRaWAN spec.
There's either an issue in encoding/decoding the payload or band package

pdubois22, 2019-03-13:
There was indeed an issue with the channels
If the NS cannot schedule more MAC that can fit in one frame in DR0, this issue can be closed.

closed 2019-05-23 by rvolosatovs