TheThingsNetwork/lorawan-stack #288: app_s_key not stored when set with cli or api

pdubois22, 2019-03-16:
**Summary:** 
When creating a device with cli or the api (as api) with the flag --session.keys.app-s-key.key=xxx, the app_s key is not stored (or not stored properly). 

This is a recent regression

When the app_s_key is derived by the JS during a JR, the key is properly stored.

**Steps to Reproduce:** 
create an ABP device or an OTAA device with app_s_key specified:

```
go run ./cmd/ttn-lw-cli  --ca="/Users/pdubois/go/src/go.thethings.network/lorawan-stack/cert.pem" devices create caf 850706c76703591801670ce770f41899 --defaults=true \
--supports-join=false \
--join-eui=70B3D59520000000 --dev-eui=0018B2000000263E \
--session.dev-addr=FC0069C7 \
--session.keys.app-s-key.key=167AB48EFF7341C4AF24138F249A845F \
--session.keys.f-nwk-s-int-key.key=B350421D287EE2202E043D341C237F3B \
--lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
--frequency-plan-id EU_863_870
```

then try to get the app_s_key value

```
go run ./cmd/ttn-lw-cli  --ca="/Users/pdubois/go/src/go.thethings.network/lorawan-stack/cert.pem" devices get caf 850706c76703591801670ce770f41899 --session 
{
  "ids": {
    "device_id": "850706c76703591801670ce770f41899",
    "application_ids": {
      "application_id": "caf"
    }
  },
  "created_at": "2019-03-16T07:24:29.362Z",
  "updated_at": "2019-03-16T07:24:29.400Z",
  "session": {
    "dev_addr": "FC0069C7",
    "keys": {
      "f_nwk_s_int_key": {
        "key": "s1BCHSh+4iAuBD00HCN/Ow=="
      },
      "s_nwk_s_int_key": {
        "key": "s1BCHSh+4iAuBD00HCN/Ow=="
      },
      "nwk_s_enc_key": {
        "key": "s1BCHSh+4iAuBD00HCN/Ow=="
      }
    },
    "started_at": "2019-03-16T07:24:29.432558Z"
  }
}

```
app_s_key is missing

this create a subsequent error for uplinks:
 `ERROR Finished unary call                      duration=189.04µs error=error:pkg/joinserver:registry_operation (registry operation failed) error_cause=error:pkg/joinserver/redis:invalid_identifiers (invalid identifiers) `

Please note that the app_s_key is ok when got from the JS during OTAA:

```
go run ./cmd/ttn-lw-cli  --ca="/Users/pdubois/go/src/go.thethings.network/lorawan-stack/cert.pem" devices create caf 850706c76703591801670ce770f41899 --defaults=true \
--supports-join=true \
--join-eui=70B3D59520000000 --dev-eui=0018B2000000263E \
--root-keys.app-key.key=EC2C176FA5CEE94CCA1EC0586C7EE0ED \
--lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
--frequency-plan-id EU_863_870

```

after Join:

```
go run ./cmd/ttn-lw-cli  --ca="/Users/pdubois/go/src/go.thethings.network/lorawan-stack/cert.pem" devices get caf 850706c76703591801670ce770f41899 --session
{
  "ids": {
    "device_id": "850706c76703591801670ce770f41899",
    "application_ids": {
      "application_id": "caf"
    },
    "dev_eui": "0018B2000000263E",
    "join_eui": "70B3D59520000000"
  },
  "created_at": "2019-03-16T07:30:04.142Z",
  "updated_at": "2019-03-16T07:30:04.177Z",
  "session": {
    "dev_addr": "FC00687A",
    "keys": {
      "session_key_id": "AWmFaJsiRtvuikeuMEaPeg==",
      "f_nwk_s_int_key": {
        "key": "nvPr1JZ2udl6lJhMIuovFQ=="
      },
      "s_nwk_s_int_key": {
        "key": "nvPr1JZ2udl6lJhMIuovFQ=="
      },
      "nwk_s_enc_key": {
        "key": "nvPr1JZ2udl6lJhMIuovFQ=="
      },
      "app_s_key": {
        "key": "KdN5txFykUvRkhuk4mEKbw=="
      }
    },
    "last_f_cnt_up": 1,
    "last_n_f_cnt_down": 1,
    "started_at": "2019-03-16T07:30:29.348485Z"
  }
}
```

**What do you see now?**
app_s_key not there.
regression. not sure when introduced. sometime between >= 03/11 and <= 04/14. 

**What do you want to see instead?**
app_s_key is required for the as

**How do you propose to implement this?**

**Environment:**
master as of 03/16

**What can you do yourself and what do you need help with?**

pdubois22, 2019-03-16:
could not reproduce it with the latest master
closing for now

johanstokking, 2019-03-16:
@pdubois22 I can reproduce this

johanstokking, 2019-03-16:
The cause is that the unexpanded `session` field mask is set for AS and then NS, where it gets overridden. This should be expanded.

PR #295 (2019-03-18): Select and merge allowed bottom level fields from devices
**Summary:**

Fixes #288 

**Changes:**

- When selecting fields from components, select the allowed bottom level fields. Before, if you select `session` from NS and AS (both allowed), when merging the fields, the NS `session` would override the AS `session`. Now, since bottom level fields are selected only, replacement of parents doesn't happen anymore
- Extend utilities for field masks for the above
- Use utilities in CLI where a local func was otherwise used
closed 2019-03-18 by johanstokking