TheThingsNetwork/lorawan-stack #289: wrong wireType = 5 for field BatteryPercentage

pdubois22, 2019-03-16:
**Summary:** 
uplink discarded because of wrong wiretype

**Steps to Reproduce:** 

**What do you see now?**
WARN Failed to match device                   dev_addr=FC0068EF error=proto: wrong wireType = 5 for field BatteryPercentage grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D633245RGZP7EV49RFZ64TT4

**What do you want to see instead?**
uplink accepted

**How do you propose to implement this?**
if this is a generation problem, that should be easy to fix.
otherwise, if the battery percentage cannot be retrieved, it should be ignored, but the rest of the uplink should be processed.

**Environment:**
master as of 03/16 12:00

**What can you do yourself and what do you need help with?**

...

johanstokking, 2019-03-16:
@pdubois22 there's a breaking change in the storage of protos; i.e. they were stored in the old format and you're using a newer version of the code.

We considered breaking storage and API changes still possible during release candidates, before the `v3.0.0` release which will be tagged Monday or Tuesday. It's very inconvenient, but it's the last opportunity to do so with V3.

closed 2019-03-16 by johanstokking

pdubois22, 2019-03-16:
fine