TheThingsNetwork/lorawan-stack #290: Stack fails to match ABP Device

Maverick10, 2019-03-17:
**Summary:** 

I registered an ABP device on TTN V3 stack and it worked fine at the beginning. Whenever I power off the device and then power it back on, the stack logs say `WARN Failed to match device`. When I delete the device and then register it again, it works fine again. And the cycle begins anew.

**Steps to Reproduce:** 

1. Register an ABP device on V3 stack
2. Power off the device and then turn it back on
3. Delete the device from registry and then register it again

**What do you see now?**

[image]

**What do you want to see instead?**

Device uplink handled.

**Environment:**

Gateway: Multitech Conduit AEP

rvolosatovs, 2019-03-18:
Does the device reset frame counters on restart?

htdvisser, 2019-03-18:
@rvolosatovs just beat me to commenting. It's indeed very likely that your device resets its frame counters to 0 on power-cycles. This is non-compliant LoRaWAN behavior, and the network server rejects messages with already-seen frame counters.

The best solution would be to use OTAA, but if you really need to use ABP, make sure to register your device with the `--resets-f-cnt` flag.

johanstokking, 2019-03-18:
@Maverick10

Maverick10, 2019-03-18:
Thanks @rvolosatovs  @htdvisser  for your replies.

I wanted to use OTAA in the first place, but I couldn't find out how to extract the AppEUI of my application. So what command do I have to write to get that?

johanstokking, 2019-03-18:
@Maverick10 you can come up with your own AppEUI.

Maverick10, 2019-03-18:
Don't I have to somehow associate this AppEUI -that I come up with- to the application?

johanstokking, 2019-03-18:
No; it's a per device setting. The AppEUI is not related to the application; in newer LoRaWAN versions it's renamed to JoinEUI because it points to a Join Server, not to an application.

closed 2019-04-04 by johanstokking