TheThingsNetwork/lorawan-stack #298: Alias EndDevice flags for LoRaWAN 1.0.x terms

htdvisser, 2019-03-18:
**Summary:** 

We should try to create flag aliases for EndDevice fields that have different names in LoRaWAN 1.0.x and LoRaWAN 1.1, such as AppEUI (instead of JoinEUI), NwkSKey instead of NwkSIntKey.

**Why do we need this?**

These flag aliases would help users transition from LoRaWAN 1.0.x to 1.1.

**What is already there? What do you see now?**

The `--join-eui` and `--nwk-s-int-key` flags.

**What is missing? What do you want to see?**

`--app-eui` and `--nwk-s-key` flags that inform (warn) users about the correct flag names, but still set those values.

**How do you propose to implement this?**

The `pflag` library doesn't seem to have builtin support for aliases, so we may have to implement this by hand.

**What can you do yourself and what do you need help with?**

I can do this.

johanstokking, 2019-03-18:
Would be an alias for **F**NwkSIntKey. The confusion is real.

htdvisser, 2019-03-18:
Updated description

PR #434 (2019-04-03): Add legacy EndDevice flags as deprecated flags

**Summary:**

This PR closes #298 by adding legacy EndDevice flags to the CLI as deprecated flags. 

These deprecated flags (such as `session.keys.nwk-s-key`) will print a warning `Flag --session.keys.nwk-s-key.key has been deprecated, use the session.keys.f_nwk_s_int_key.key flag` and forward the value to the new flags until the user learns new habits.
closed 2019-04-03 by htdvisser